supermerill/SuperSlicer #2534: Donate button or menu item

SamSpiri, 2022-03-14:
Hi,
Loving the project an wish you all the best.
Please put a donation options on the top level in the menu or a button always visible in the app window.
Patreon and/or paypal would be better option for users who don't know about github.
I wish you could pay contributors and get a good profit yourself.
Cheers!

ModMike, 2022-03-21:
Where did you find the link?

SamSpiri, 2022-03-22:
when you go to home page of the project on github, look on the right column in the middle of it.

SamSpiri, 2022-03-22:
@supermerill Hey man, we can't find how to donate really, we are lamer users ;)

supermerill, 2022-03-22:
[image]

ModMike, 2022-03-24:
Never donated through GitHub before. Enjoy the donation!