supermerill/SuperSlicer #2535: Z-seam aligned

tome9111991, 2022-03-14:
Hi
I use almost aligned for Z seam.
But since susi .58 i have some issue with.
I have many random seams looks not really aligend.
(Sorry dont have a screenshot yet)
You can look in 3D benchy the seam on back looks a bit random.
In PS2.4 looks a bit "cleaner"
But best result i have in susi .57

Mayve you can look into.
Thx

supermerill, 2022-03-14:
kinda same bag as #2444

Deutherius, 2022-04-04:
I have also came across this bug. Beginning with 2.4, I do not think it is related to #2444, which seams to be focused on seam painting. I have an issue with aligned when no painting is used.

Aligned z seam seems to be broken/very unreliable. Where it was previously reliable in that it chose some position and made the seam truly aligned (z seam gap artificially enlarged for better visibility)

[image]

it will now place the seam randomly

[image]

Furthermore, even if it does place the seam aligned, it will sometimes choose a really weird position, i.e. in the middle of a wall when a suitable corner is nearbly. Example from 2.3.57.9

[image]

and the same object in 2.4.58.2

[image]

Same profiles were used for both versions of superslicer (with the exception of z seam gap, but I have verified that this has no effect on seam placement). Included 3MFs for ease of checking:
[SB_front_and_latch.zip](https://github.com/supermerill/SuperSlicer/files/8412510/SB_front_and_latch.zip)

supermerill, 2022-04-28:
seems okay-ish in the nightly.
A final problem is that it seems that the "aligned" may make the seam to jump a little bit on the previous/next point sometimes. Will look why.

closed 2022-07-21 by tome9111991