supermerill/SuperSlicer #2541: Bottom layer above supports is only one layer thick in certain areas

coldyoungguy, 2022-03-15:
### What happened?

When slicing the bottom layer above the supports is only one layer thick when i have set it to 5

some photos demonstrating the issue:
The following photos are one layer apart
[image]
[image]
[image]

As can be seen from the screenshots, the inner circle requiring supports generated the 5 layers correctly but the outer circle is only one layer thick

### Project file & How to reproduce

Not much steps to reproduce the bug
[LidApplicator.zip](https://github.com/supermerill/SuperSlicer/files/8254630/LidApplicator.zip)

### Version

2.3.57 and 2.4.58

### Operating system

Windows 11

### Printer model

Voron 2.4 with Afterburner

neophyl, 2022-03-15:
Change the setting called 'No solid infill over'  Its currently set to 2.  Its a fairly recent setting to try and combat extra solid infill on sloping walls.  I think its interpreting your flat raised section and limiting it to 2 layers.  When I set it to 5 or more I get the 5 bottom layers.

coldyoungguy, 2022-03-21:
> Change the setting called 'No solid infill over' Its currently set to 2. Its a fairly recent setting to try and combat extra solid infill on sloping walls. I think its interpreting your flat raised section and limiting it to 2 layers. When I set it to 5 or more I get the 5 bottom layers.

I see, i unchecked the "Ensure vertical shell thickness" and it managed to slice with 5 bottoms layers, thanks!

i guess its a feature not a bug haha

supermerill, 2022-03-22:
What was going on (not 100% sure):
1 - the "add extra perimeters On overhangs" change the layer over supports from '3 perimeters & bridges' to 'only perimeters'
2 - the "keep only bridges" removes all these overhangs, 'removing' that layer'
3 - the 'no solid over perimeter' see these overhangs, and  remove soid infill from 2 layers above.
4 - now there are only one solid layer between the now erased overhangs and where the 'no solid over perimeter' kicks in.

I advise to not use 'keep only bridges' and 'extra peiemters on ovarhangs', as the first erase the second. I may have to add a warning message.

lightoze, 2022-05-06:
I think I encountered a similar problem. Overhang/support settings are all from system profile, and here's what I get for one model (no solid infill at all, when profile is set to 5):
[image]

The only way I managed to get it to work was setting "no solid infill over" to 0 perimeters. "No perimeters on bridge areas" is disabled and changing it does not improve anything.

Exactly same settings on another model correctly generate 4 solid layers over one bridge infill layer:
[image]