supermerill/SuperSlicer #2543: Crash on OSX

spuder, 2022-03-15:
### What happened?

Downloaded the latest pre-release of super slicer for OSX. As soon as I opened it crashed. (Running on M1 Mac, but should still work under Intel/AMD emulation)

```
-------------------------------------
Translated Report (Full Report Below)
-------------------------------------

Process:               SuperSlicer [68144]
Path:                  /Applications/SuperSlicer.app/Contents/MacOS/SuperSlicer
Identifier:            com.merill.slic3r/
Version:               SuperSlicer SuperSlicer_2.4.58.1_2022-03-07 (SuperSlicer_2.4.58.1_2022-03-07)
Code Type:             X86-64 (Translated)
Parent Process:        launchd [1]
User ID:               501

Date/Time:             2022-03-15 17:29:09.2672 -0600
OS Version:            macOS 12.2.1 (21D62)
Report Version:        12
Anonymous UUID:        C1CE877F-9609-0BE6-2576-DB59E07701EC

Sleep/Wake UUID:       7814BFF3-993A-4E6B-BEBF-4BFF91E04D56

Time Awake Since Boot: 180000 seconds
Time Since Wake:       2893 seconds

System Integrity Protection: enabled

Crashed Thread:        0

Exception Type:        EXC_CRASH (SIGABRT)
Exception Codes:       0x0000000000000000, 0x0000000000000000
Exception Note:        EXC_CORPSE_NOTIFY

Termination Reason:    Namespace DYLD, Code 1 Library missing
Library not loaded: /usr/local/opt/zstd/lib/libzstd.1.dylib
Referenced from: /Applications/SuperSlicer.app/Contents/MacOS/SuperSlicer
Reason: tried: '/usr/local/opt/zstd/lib/libzstd.1.dylib' (no such file), '/usr/local/lib/libzstd.1.dylib' (no such file), '/usr/lib/libzstd.1.dylib' (no such file)
(terminated at launch; ignore backtrace)

Application Specific Information:
Library not loaded: /usr/local/opt/zstd/lib/libzstd.1.dylib
Referenced from: /Applications/SuperSlicer.app/Contents/MacOS/SuperSlicer
Reason: tried: '/usr/local/opt/zstd/lib/libzstd.1.dylib' (no such file), '/usr/local/lib/libzstd.1.dylib' (no such file), '/usr/lib/libzstd.1.dylib' (no such file)

Thread 0 Crashed:
0   Rosetta Runtime Routines	       0x10a60d940 ???
1   <translation info unavailable>	       0x10a654eb4 ???
2   dyld                          	       0x2073b2201 abort_with_payload + 9
3   dyld                          	       0x207360613 dyld4::halt(char const*) + 375
4   dyld                          	       0x20735c55d dyld4::prepare(dyld4::APIs&, dyld3::MachOAnalyzer const*) + 3955
5   dyld                          	       0x20735b4b4 start + 388

Thread 1:: com.apple.rosetta.exceptionserver
0   runtime                       	    0x7ff7ffd1f8e4 0x7ff7ffd1b000 + 18660
1   runtime                       	    0x7ff7ffd2c928 0x7ff7ffd1b000 + 71976
2   runtime                       	    0x7ff7ffd2e0a4 0x7ff7ffd1b000 + 77988

Thread 0 crashed with ARM Thread State (64-bit):
    x0: 0x0000000000000006   x1: 0x0000000000000001   x2: 0x000000030f8cfad0   x3: 0x0000000000000075
    x4: 0x000000030f8cf6d0   x5: 0x0000000000000000   x6: 0xe5c4287345f600c3   x7: 0x000000200f8cf6e0
    x8: 0x000000030f8cf6d0   x9: 0x0000000000000000  x10: 0x0000000000000075  x11: 0x000000030f8cf6d0
   x12: 0x0000000000000075  x13: 0x000000030f8cfad0  x14: 0x0000000000000001  x15: 0x0000000000000006
   x16: 0x0000000000000209  x17: 0x0000000000000000  x18: 0x8000000112676b70  x19: 0x000000010a603000
   x20: 0x00000000000007ff  x21: 0x000000011267bfa0  x22: 0xffffffffffffffff  x23: 0x000000010a60d914
   x24: 0x00000002073b21cf  x25: 0x000000010a60d1f4  x26: 0x000000010a6534e8  x27: 0x0000000000000000
   x28: 0x000000020739c0ce   fp: 0x000000002c000000   lr: 0x000000010a654eb4
    sp: 0x0000000112676b70   pc: 0x000000010a60d940 cpsr: 0x80001000
   far: 0x0000000112c35010  esr: 0x56000080  Address size fault
[... 182 lines cut ...]
Display: Color LCD, 3456 x 2234 Retina, Main, MirrorOff, Online
Memory Module: LPDDR5
AirPort: Wi-Fi, wl0: Dec 13 2021 17:12:34 version 20.10.853.31.8.7.109 FWID 01-804e92d8
Bluetooth: Version (null), 0 services, 0 devices, 0 incoming serial ports
Network Service: Wi-Fi, AirPort, en0
USB Device: USB31Bus
USB Device: USB31Bus
USB Device: USB31Bus
Thunderbolt Bus: MacBook Pro, Apple Inc.
Thunderbolt Bus: MacBook Pro, Apple Inc.
Thunderbolt Bus: MacBook Pro, Apple Inc.

```

### Project file & How to reproduce

Download and open app on OSX

### Version

SuperSlicer_2.4.58.1_2022-03-07

### Operating system

OSX

liftbag, 2022-03-16:
See the #2443 and the #2488

foxmckraut, 2022-03-16:
got the same problem and manualy installed zstd like in the log but there is another problem. 

`-------------------------------------
Translated Report (Full Report Below)
-------------------------------------

Process:               SuperSlicer [1720]
Path:                  /Users/USER/Desktop/SuperSlicer.app/Contents/MacOS/SuperSlicer
Identifier:            com.merill.slic3r/
Version:               SuperSlicer SuperSlicer_2.4.58.1_2022-03-07 (SuperSlicer_2.4.58.1_2022-03-07)
Code Type:             X86-64 (Translated)
Parent Process:        launchd [1]
User ID:               501

Date/Time:             2022-03-16 09:06:53.5258 +0100
OS Version:            macOS 12.2.1 (21D62)
Report Version:        12
Anonymous UUID:        E2AD3586-C427-4A05-B447-BC5FA38B32A4

Sleep/Wake UUID:       E36F6562-0D3A-4A6B-B745-84B05AB8F3AA

Time Awake Since Boot: 1600 seconds
Time Since Wake:       1041 seconds

System Integrity Protection: enabled

Crashed Thread:        0

Exception Type:        EXC_CRASH (SIGABRT)
Exception Codes:       0x0000000000000000, 0x0000000000000000
Exception Note:        EXC_CORPSE_NOTIFY

Termination Reason:    Namespace DYLD, Code 1 Library missing
Library not loaded: /usr/local/opt/zstd/lib/libzstd.1.dylib
Referenced from: /Users/USER/Desktop/SuperSlicer.app/Contents/MacOS/SuperSlicer
Reason: tried: '/usr/local/opt/zstd/lib/libzstd.1.dylib' (mach-o file, but is an incompatible architecture (have 'arm64', need 'x86_64')), '/usr/local/lib/libzstd.1.dylib' (no such file), '/usr/lib/libzstd.1.dylib' (no such file), '/usr/local/opt/zstd/lib/libzstd.1.5.2.dylib' (mach-o file, but is an incompatible architecture (have 'arm64', need 'x86_64')), '/usr/local/lib/libzstd.1.5.2.dylib' (no such file), '/usr/lib/libzstd.1.5.2.dylib' (no such file)
(terminated at launch; ignore backtrace)

Application Specific Information:
Library not loaded: /usr/local/opt/zstd/lib/libzstd.1.dylib
Referenced from: /Users/gordonkretzschmar/Desktop/SuperSlicer.app/Contents/MacOS/SuperSlicer
Reason: tried: '/usr/local/opt/zstd/lib/libzstd.1.dylib' (mach-o file, but is an incompatible architecture (have 'arm64', need 'x86_64')), '/usr/local/lib/libzstd.1.dylib' (no such file), '/usr/lib/libzstd.1.dylib' (no such file), '/usr/local/opt/zstd/lib/libzstd.1.5.2.dylib' (mach-o file, but is an incompatible architecture (have 'arm64', need 'x86_64')), '/usr/local/lib/libzstd.1.5.2.dylib' (no such file), '/usr/lib/libzstd.1.5.2.dylib' (no such file)

Thread 0 Crashed:
0   Rosetta Runtime Routines	       0x1062b6940 ???
1   <translation info unavailable>	       0x106300ddc ???
2   dyld                          	       0x202ebf201 abort_with_payload + 9
3   dyld                          	       0x202e6d613 dyld4::halt(char const*) + 375
4   dyld                          	       0x202e6955d dyld4::prepare(dyld4::APIs&, dyld3::MachOAnalyzer const*) + 3955
5   dyld                          	       0x202e684b4 start + 388

Thread 1:: com.apple.rosetta.exceptionserver
0   runtime                       	    0x7ff7ffc358e4 0x7ff7ffc31000 + 18660
1   runtime                       	    0x7ff7ffc42928 0x7ff7ffc31000 + 71976
2   runtime                       	    0x7ff7ffc440a4 0x7ff7ffc31000 + 77988

Thread 0 crashed with ARM Thread State (64-bit):
    x0: 0x0000000000000006   x1: 0x0000000000000001   x2: 0x0000000307085aa0   x3: 0x0000000000000088
    x4: 0x00000003070856a0   x5: 0x0000000000000000   x6: 0x0000002000000000   x7: 0x0000002007086058
    x8: 0x00000003070856a0   x9: 0x0000000000000000  x10: 0x0000000000000088  x11: 0x00000003070856a0
   x12: 0x0000000000000088  x13: 0x0000000307085aa0  x14: 0x0000000000000001  x15: 0x0000000000000006
   x16: 0x0000000000000209  x17: 0x0000000000000000  x18: 0x800000010e31fb70  x19: 0x00000001062ac000
   x20: 0x00000000000007ff  x21: 0x000000010e324fa0  x22: 0xffffffffffffffff  x23: 0x00000001062b6914
   x24: 0x0000000202ebf1cf  x25: 0x00000001062b61f4  x26: 0x00000001062ff410  x27: 0x0000000000000000
   x28: 0x0000000202ea90ce   fp: 0x000000002c000000   lr: 0x0000000106300ddc
    sp: 0x000000010e31fb70   pc: 0x00000001062b6940 cpsr: 0x80001000
   far: 0x000000010e8de010  esr: 0x56000080  Address size fault

Binary Images:
               0x0 - 0xffffffffffffffff ??? (*) <00000000-0000-0000-0000-000000000000> ???
[... 164 lines cut ...]
      "rolloutId" : "5ffde50ce2aacd000d47a95f",
      "factorPackIds" : {

      },
      "deploymentId" : 240000123
    }
  ],
  "experiments" : [

  ]
}
}

Model: MacBookAir10,1, BootROM 7429.81.3, proc 8:4:4 processors, 8 GB, SMC 
Graphics: Apple M1, Apple M1, Built-In
Display: Color LCD, 2560 x 1600 Retina, Main, MirrorOff, Online
Memory Module: LPDDR4
AirPort: Wi-Fi, wl0: Oct 26 2021 16:10:41 version 18.20.310.15.7.8.120 FWID 01-5d6c3867
Bluetooth: Version (null), 0 services, 0 devices, 0 incoming serial ports
Network Service: Wi-Fi, AirPort, en0
USB Device: USB31Bus
USB Device: USB31Bus
Thunderbolt Bus: MacBook Air, Apple Inc.
Thunderbolt Bus: MacBook Air, Apple Inc.
`

supermerill, 2022-04-10:
can you test this build for macos arm? https://mega.nz/file/9gwyHYKS#-BJPvMGzupZPLbnMqWBF3elMV7BLzXXqnBZ6IXNO2HU
I tried with zstd uninstalled, so it shouldn't be needed, but I'm not 100% confident.

K2Kevin, 2022-04-15:
Same on Catalina, Crash at startup for all versions 2.4.xxx

alexvonsiebenthal, 2022-04-16:
> can you test this build for macos arm? [mega.nz/file/9gwyHYKS#-BJPvMGzupZPLbnMqWBF3elMV7BLzXXqnBZ6IXNO2HU](https://mega.nz/file/9gwyHYKS#-BJPvMGzupZPLbnMqWBF3elMV7BLzXXqnBZ6IXNO2HU) I tried with zstd uninstalled, so it shouldn't be needed, but I'm not 100% confident. i guess it still need the xattr thingy to be launch on another computer than the one that built it?

This version is flagged as damaged and won't be opened by macOS

foxmckraut, 2022-04-16:
> This version is flagged as damaged and won't be opened by macOS

try:
sudo xattr -cr "/Volumes/MacOS SSD/Applications/SuperSlicer.app"
or

sudo xattr -cr "/Volumes/Macintosh HD/Applications/SuperSlicer.app"

ohheyrj, 2022-04-30:
> > > can you test this build for macos arm? [mega.nz/file/9gwyHYKS#-BJPvMGzupZPLbnMqWBF3elMV7BLzXXqnBZ6IXNO2HU](https://mega.nz/file/9gwyHYKS#-BJPvMGzupZPLbnMqWBF3elMV7BLzXXqnBZ6IXNO2HU) I tried with zstd uninstalled, so it shouldn't be needed, but I'm not 100% confident. i guess it still need the xattr thingy to be launch on another computer than the one that built it?
> > 
> > 
> > This version is flagged as damaged and won't be opened by macOS
> 
> try: sudo xattr -cr "/Volumes/MacOS SSD/Applications/SuperSlicer.app" or
> 
> sudo xattr -cr "/Volumes/Macintosh HD/Applications/SuperSlicer.app"
> 
> in terminal

Just to confirm, these steps worked for me, will test the beta now :)