supermerill/SuperSlicer #2545: Incorrect jerk units

jullit31, 2022-03-16:
**Bug description**
Minor thing, but the jerk is expressed in mm/s when it should be mm/s³.

**To Reproduce**
Steps to reproduce the behavior:
1. Go to 'Machine limits'
2. Scroll down to 'Jerk limits'
3. See error

**Screenshots**
[image]

**Desktop:**
 - OS:  Windows
 - Version 2.3.57.11

jullit31, 2022-03-16:
Hmm, interesting. So it's a mistake in the Marlin docs. I mean, the physics is clear: jerk is the rate of change of acceleration, so it must be of the dimension L/T³ (length over time cubed).

P.S. Having the same units in the GUI for all G-code flavors is also easier for the user IMHO, so good choice 👍

CCS86, 2022-03-23:
I don't think any of the main printer firmwares use "true" jerk. It is just an acceleration "bypass", where the commanded velocity can jump instantaneously up by the max jerk value, continuing from there bounded by acceleration limits.

closed 2022-09-09 by jullit31