supermerill/SuperSlicer #2546: MacOS 12.3 removed python command, replaced by python3

bistory, 2022-03-16:
### What happened?

MacOS 12.3 has been released and it uses python3 as default BUT not with the usual "python" command.
So post-processing scripts are not executed anymore.

Here is the error I get : 

```
Post-processing script /Users/thomaslete/Projects/SuperSlicer/FastSQV.py on file /Users/thomaslete/Downloads/Duo_Cartridge+LID.gcode failed.
Error code: 127
Output:
zsh:1: /Users/thomaslete/Projects/SuperSlicer/FastSQV.py: bad interpreter: python: no such file or directory
```

Thanks for your help :-)

### Project file & How to reproduce

Use this project for gcode post-processing : https://github.com/K-Tett/Klipper-FastSQV

Here are some clues : https://macmule.com/2022/01/29/macos-monterey-12-3-will-remove-python-2-7-usr-bin-python/

### Version

2.3.57.11

### Operating system

MacOS 12.3

### Printer model

Voron 2.4

supermerill, 2022-03-16:
then replace the `python` from your post-processing command by `python3` ?

bistory, 2022-03-16:
Ok I feel dumb now, I thought it was SuperSlicer that was calling python 🙄
Thanks ...!

closed 2022-03-16 by bistory