supermerill/SuperSlicer #2552: Skinny Dip comma instead of dot

foxmckraut, 2022-03-19:
### What happened?

if you activate Skinnydip it puts this in your gcode and stops at this state.

; SKINNYDIP START
M73 P11 R50
G1 E31,0000 F1980
G4 P0
M73 P11 R50
G1 E-31,0000 F4200
G4 P0
; SKINNYDIP END
M73 P12 R50
G1 E-50.0000 F2000
G1  Y166.285  F2400
G4 S0
; Filament-specific end gcode 
;END gcode for filament

g1 e31,0000 is not a valid gcode

### Project file & How to reproduce

filament settings > multimaterial > enable skinny dipping > slice.

### Version

2.4.58.1

### Operating system

Windows10 and macosx 12.2.1

### Printer model

Anycubic i3 mega s

supermerill, 2022-03-23:
In what language is the os (just to be sure) ?

foxmckraut, 2022-03-23:
> In what language is the os (just to be sure) ?

both are german.

supermerill, 2022-03-23:
fixed for next nightly

closed 2022-07-14 by supermerill