supermerill/SuperSlicer #2557: Open Infill for Epoxy Filling

JoeLoginIsAlreadyTaken, 2022-03-21:
I'm looking for a way to create a open line grid pattern with SuperSlicer.
I use this for prints which should be filled with epoxy (just print the outer shell and fill it with resin and maybe sand).
As the pattern is open in X+Y direction, the epoxy can flow in every corner.

In Cura i used the "Lines" infill-pattern with custom directions "[45,45,45,135,135,135]".
Always 3 Layers with lines in the same direction, then 3 layers with lines in the opposite direction (at 0.2mm layers).
The infill have keep the walls in position and prevent them from deforming, as the epoxy causes some pressure.

It would be very nice if SuperSlicer could make this possible
[image]
[image]
[image]
.

neophyl, 2022-03-22:
With the release version, just select an infill pattern like grid and then set the increment value to something like 10-15 (Print Settings>Infill>Advanced>Angle>Increment.  Although many of the patterns would work with suitable increment angles.

Those settings are Expert/Advanced level so make sure you aren't in Simple mode.

supermerill, 2022-03-22:
gyroid is open by construction.
I guess i can create  the same kind of array for direction, instead of only one value.

closed 2024-02-11 by supermerill