supermerill/SuperSlicer #2561: filament_shrink no effect

Ajol3D, 2022-03-23:
### What happened?

Hi, 
**filament_shrink parameter seems have no effect on gcode.**

I noticed that when I tried to correct the shrinkage on ASA filament, my parts were still the same size despite having changed the parameter. Maybe I didn't understand the use of this parameter?

### Project file & How to reproduce

1. Apply 100% to filament_shrink
[image]
[image]

2. Apply 90% to filament_shrink
[image]
[image]

-> In all cases on line 59194 of gcode x pos is always 104.14

### Version

2.4.58.1

### Operating system

Windows 10

### Printer model

Sapphire Plus

neophyl, 2022-03-23:
https://github.com/supermerill/SuperSlicer/issues/2517

Ajol3D, 2022-03-23:
Ok, Thanks

closed 2022-03-23 by Ajol3D