supermerill/SuperSlicer #2564: SuperSlicer (2.4.58.0) crashes on startup

msrd0, 2022-03-23:
### What happened?

When starting superslicer, it opens the loading screen, then the main screen and then immediately crashes. This is the backtrace:

```
#0  0x00007ffff27e934c in __pthread_kill_implementation () at /usr/lib/libc.so.6
#1  0x00007ffff279c4b8 in raise () at /usr/lib/libc.so.6
#2  0x00007ffff2786534 in abort () at /usr/lib/libc.so.6
#3  0x00007ffff7df17ee in __gnu_cxx::__verbose_terminate_handler() () at /usr/src/debug/gcc/libstdc++-v3/libsupc++/vterminate.cc:95
#4  0x00007ffff7dfdc4c in __cxxabiv1::__terminate(void (*)()) (handler=<optimized out>) at /usr/src/debug/gcc/libstdc++-v3/libsupc++/eh_terminate.cc:48
#5  0x00007ffff7dfdcb9 in std::terminate() () at /usr/src/debug/gcc/libstdc++-v3/libsupc++/eh_terminate.cc:58
#6  0x0000555556008e64 in Slic3r::GUI::Plater::priv::~priv() ()
#7  0x000055555603bf91 in Slic3r::GUI::Plater::~Plater() ()
#8  0x000055555603bfc9 in Slic3r::GUI::Plater::~Plater() ()
#9  0x00007ffff39f767b in wxBookCtrlBase::DeletePage(unsigned long) (this=0x555557c600a0, nPage=<optimized out>) at ./src/common/bookctrl.cpp:384
#10 0x00007ffff39b162f in wxNotebook::DeleteAllPages() (this=0x555557c600a0) at ./src/gtk/notebook.cpp:413
#11 0x00007ffff39b051b in wxNotebook::~wxNotebook() (this=0x555557c600a0) at ./src/gtk/notebook.cpp:163
#12 0x00007ffff39b05f9 in wxNotebook::~wxNotebook() (this=0x555557c600a0) at ./src/gtk/notebook.cpp:155
#13 0x00007ffff3ae310f in wxWindowBase::Destroy() (this=0x555557c600a0) at ./src/common/wincmn.cpp:568
#14 wxWindowBase::DestroyChildren() (this=0x555557a54190) at ./src/common/wincmn.cpp:600
#15 0x00007ffff395961b in wxWindow::~wxWindow() (this=0x555557a54190) at ./src/gtk/window.cpp:2750
#16 0x0000555555ff0029 in Slic3r::GUI::MainFrame::~MainFrame() ()
#17 0x00007ffff39f10b4 in wxAppBase::DeleteAllTLWs() (this=<optimized out>) at ./src/common/appcmn.cpp:137
#18 wxAppBase::CleanUp() (this=0x555556f0ca30) at ./src/common/appcmn.cpp:150
#19 0x00007ffff3e83e52 in wxEntryCleanup() () at ./src/common/init.cpp:448
#20 0x00007ffff3e83ff5 in wxUninitialize() () at ./src/common/init.cpp:565
#21 wxInitializer::~wxInitializer() (this=0x7fffffff57c0) at ./include/wx/init.h:109
#22 wxEntry(int&, wchar_t**) (argc=@0x7ffff3fd5218: 1, argv=<optimized out>) at ./src/common/init.cpp:507
#23 0x0000555555f9e0bd in Slic3r::GUI::GUI_Run(Slic3r::GUI::GUI_InitParams&) ()
#24 0x0000555555780125 in Slic3r::CLI::run(int, char**) ()
#25 0x0000555555782d38 in main ()
```

### Project file & How to reproduce

Just start superslicer

### Version

2.4.58.0

### Operating system

ArchLinux 5.16.16-zen1-1-zen

supermerill, 2022-03-23:
The 2.4.58.0 has many bugs.
Can you try a more recent one, like the 2.5.58.2, from a nightly? (note that i don't know what/how you got it, so i assume it's an appimage from github)
(https://github.com/supermerill/SuperSlicer/suites/5761699031/artifacts/191585336)

msrd0, 2022-03-23:
I get the same problem with 2.4.58.1 (I compiled it from source), I somehow must've missed that there was a newer tag before. I also tried compiling the master branch but it failed with some linker error.

supermerill, 2022-03-23:
not very helpful, as the stacktrace show an error after the crash, hiding the problem.
Did you compile whith debug flag?

msrd0, 2022-03-23:
yes I compiled with with debug flag in release mode (`-O2 -g`)

msrd0, 2022-03-23:
I managed to get a more complete backtrace:

```
#0  0x00007ffff27e934c in __pthread_kill_implementation () at /usr/lib/libc.so.6
#1  0x00007ffff279c4b8 in raise () at /usr/lib/libc.so.6
#2  0x00007ffff2786534 in abort () at /usr/lib/libc.so.6
#3  0x00007ffff7df17ee in __gnu_cxx::__verbose_terminate_handler() () at /usr/src/debug/gcc/libstdc++-v3/libsupc++/vterminate.cc:95
#4  0x00007ffff7dfdc4c in __cxxabiv1::__terminate(void (*)()) (handler=<optimized out>) at /usr/src/debug/gcc/libstdc++-v3/libsupc++/eh_terminate.cc:48
#5  0x00007ffff7dfdcb9 in std::terminate() () at /usr/src/debug/gcc/libstdc++-v3/libsupc++/eh_terminate.cc:58
#6  0x0000555555fa3844 in std::thread::~thread() (this=0x555557ca2690) at /usr/bin/../lib64/gcc/x86_64-pc-linux-gnu/11.2.0/../../../../include/c++/11.2.0/bits/std_thread.h:152
#7  Slic3r::GUI::Mouse3DController::~Mouse3DController() (this=0x555557ca2510) at /usr/src/debug/SuperSlicer-nightly_dev/src/slic3r/GUI/Mouse3DController.hpp:29
#8  Slic3r::GUI::Plater::priv::~priv() (this=0x555557c9cfa0) at /usr/src/debug/SuperSlicer-nightly_dev/src/slic3r/GUI/Plater.cpp:2275
#9  0x0000555555fd64ec in std::default_delete<Slic3r::GUI::Plater::priv>::operator()(Slic3r::GUI::Plater::priv*) const (this=0x555557b06f88, __ptr=0x555557c9cfa0) at /usr/bin/../lib64/gcc/x86_64-pc-linux-gnu/11.2.0/../../../../include/c++/11.2.0/bits/unique_ptr.h:85
#10 std::unique_ptr<Slic3r::GUI::Plater::priv, std::default_delete<Slic3r::GUI::Plater::priv> >::~unique_ptr() (this=0x555557b06f88) at /usr/bin/../lib64/gcc/x86_64-pc-linux-gnu/11.2.0/../../../../include/c++/11.2.0/bits/unique_ptr.h:361
#11 Slic3r::GUI::Plater::~Plater() (this=0x555557b06cd0) at /usr/src/debug/SuperSlicer-nightly_dev/src/slic3r/GUI/Plater.hpp:146
#12 0x0000555555fd651d in Slic3r::GUI::Plater::~Plater() (this=0x555557b06cd0) at /usr/src/debug/SuperSlicer-nightly_dev/src/slic3r/GUI/Plater.hpp:146
#13 0x00007ffff39f767b in wxBookCtrlBase::DeletePage(unsigned long) (this=0x555557c12000, nPage=<optimized out>) at ./src/common/bookctrl.cpp:384
#14 0x00007ffff39b162f in wxNotebook::DeleteAllPages() (this=0x555557c12000) at ./src/gtk/notebook.cpp:413
#15 0x00007ffff39b051b in wxNotebook::~wxNotebook() (this=0x555557c12000) at ./src/gtk/notebook.cpp:163
#16 0x00007ffff39b05f9 in wxNotebook::~wxNotebook() (this=0x555557c12000) at ./src/gtk/notebook.cpp:155
#17 0x00007ffff3ae310f in wxWindowBase::Destroy() (this=0x555557c12000) at ./src/common/wincmn.cpp:568
#18 wxWindowBase::DestroyChildren() (this=0x5555574a0190) at ./src/common/wincmn.cpp:600
#19 0x00007ffff395961b in wxWindow::~wxWindow() (this=0x5555574a0190) at ./src/gtk/window.cpp:2750
#20 0x0000555555f8bb9d in Slic3r::GUI::MainFrame::~MainFrame() (this=0x5555574a0190) at /usr/src/debug/SuperSlicer-nightly_dev/src/slic3r/GUI/MainFrame.hpp:163
#21 0x00007ffff39f10b4 in wxAppBase::DeleteAllTLWs() (this=<optimized out>) at ./src/common/appcmn.cpp:137
#22 wxAppBase::CleanUp() (this=0x555556ef5c50) at ./src/common/appcmn.cpp:150
#23 0x00007ffff3e83e52 in wxEntryCleanup() () at ./src/common/init.cpp:448
#24 0x00007ffff3e83ff5 in wxUninitialize() () at ./src/common/init.cpp:565
#25 wxInitializer::~wxInitializer() (this=0x7fffffff5490) at ./include/wx/init.h:109
#26 wxEntry(int&, wchar_t**) (argc=@0x7ffff3fd5218: 1, argv=<optimized out>) at ./src/common/init.cpp:507
#27 0x0000555555f39959 in Slic3r::GUI::GUI_Run(Slic3r::GUI::GUI_InitParams&) (params=...) at /usr/src/debug/SuperSlicer-nightly_dev/src/slic3r/GUI/GUI_Init.cpp:54
#28 0x0000555555774b97 in Slic3r::CLI::run(int, char**) (this=<optimized out>, argc=1, argv=0x7fffffffdf98) at /usr/src/debug/SuperSlicer-nightly_dev/src/PrusaSlicer.cpp:625
#29 0x0000555555778bee in main(int, char**) (argc=351389, argv=0x6) at /usr/src/debug/SuperSlicer-nightly_dev/src/PrusaSlicer.cpp:858

supermerill, 2022-03-23:
sadly, i don't think this stacktrace is the useful one (unless you have a 3D mouse).
what's going on is probably an execption is a thread that makes the app terminate. What it's displayed here is the stacktrace of the last thread to be killed (the gui one).

Ideally, can you launch it in a debugger and makes it to stop at the first exception raised?

supermerill, 2022-03-23:
It can't access the `resources` directory, that shoudl be next to the executable, or can't create/get the user directory.
Are you sure you packaged it right after compiling?

msrd0, 2022-03-23:
Yes, I used `ninja install` and enabled `SLIC3R_FHS` (that worked perfectly for the non-prerelease) and it's installed them to `/usr/share/SuperSlicer/`, so the resources should be there. And there's nothing weird about my user directory, superslicer should be able to create its config files with no problem.

supermerill, 2022-04-02:
now, it should show an error at startup, but no crash anymore.

supermerill, 2022-04-03:
can you send me your (zipped) config dir, so i can reproduce?

msrd0, 2022-04-03:
It now shows the error message and then continues with the initial configuration dialog eventhough the config directory existed already (looks like it was created Nov 2020, so maybe it was too old? I'd have used the stable releases mostly). Here's my config directory after I launched superslicer, but the error message still appears: https://gigamove.rwth-aachen.de/de/download/447ab6eca39a76b4d3e5e4f1afacfd7a (for some reason you can't upload that to GitHub)

Also, after clicking through the configuration dialog (without changing any options), I'm unable to finish it. Is this intended?

[image]

Lukas1818, 2022-04-03:
[SuperSlicer-alpha.zip](https://github.com/supermerill/SuperSlicer/files/8404237/SuperSlicer-alpha.zip)

The message does also appear at an empty config. But it does not appear again after the wizard.

Lukas1818, 2022-04-03:
> Also, after clicking through the configuration dialog (without changing any options), I'm unable to finish it. Is this intended?

I have the same issue. But if you select `Define a custom printer profile` at `Custom Printer` you can finish the wizard.