supermerill/SuperSlicer #2566: wxWidgets Debug Alert

Imburr, 2022-03-23:
### What happened?

When I use the G> button to send my slicer g-code directly to Klipper, there is an error dialog: https://i.imgur.com/VBu4Bxp.png

The expected result is that SuperSlicer send the g-code to Klipper directly whether it is printing or not.

If I load this project into Prusa Slicer and send to Klipper, I do not get this error dialog.

### Project file & How to reproduce

This only happens if there is a file loaded into the top File window box in Klipper: https://i.imgur.com/gMFt0lE.png

If I clear this file, and then click the G>, it loads without error. If I export the g-code, and add it manually to Klipper, no error.

On the error dialog, if I click No (to not close the program), the error dialog goes away and the file is still sent to Klipper. Alternatively if I am not printing, and I click "Clear File" in Klipper, then I can send the file over without error dialog.

[Razor Spatula Cover and Mount.zip](https://github.com/supermerill/SuperSlicer/files/8335062/Razor.Spatula.Cover.and.Mount.zip)

### Version

2.4.58.1

### Operating system

Windows 11

### Printer model

Creality Ender 3 V2 4.2.2

Imburr, 2022-03-23:
> thanks for the report, maybe it will be a bit easier for me to reproduce.
> 
> duplicate of #2512 and #2492 btw

Thanks!

matthewlloyd, 2022-03-23:
Also dup of https://github.com/supermerill/SuperSlicer/issues/2396

supermerill, 2022-03-26:
can you test with this debug build?
It should emit the error the same way, but the error message should be better.
Be careful to slice a very little & simple object, as it's very slow.
https://github.com/supermerill/SuperSlicer/suites/5802201336/artifacts/193999699

Imburr, 2022-03-27:
When I run it I am presented with a string of errors:

https://i.imgur.com/NXxhqgZ.png
https://i.imgur.com/45ZpmHq.png

matthewlloyd, 2022-03-27:
Assuming it's the same bug, there's a stack trace from a debug build in my original issue here: https://github.com/supermerill/SuperSlicer/issues/2396. If it isn't the same bug, please ignore this comment.

supermerill, 2022-03-31:
can someone test this nightly? https://github.com/supermerill/SuperSlicer/suites/5835274444/artifacts/196041452
it should resolve the issue.

Imburr, 2022-03-31:
I tested, and uploaded to Klipper while printing another design two times, and no errors. At this point, the problem does look fixed. Thank you for your work and attention on this matter, I truly appreciate it as I am sure everyone who uses SS does!

closed 2022-07-14 by supermerill