supermerill/SuperSlicer #2569: 2.4.58.x: Strange Defect in Perimeter Wall

CCS86, 2022-03-23:
### What happened?

In a very uniform, in one specific spot, of one specific layer, a very strange defect is generated that affects 2 walls. One wall has a tiny kink inwards, and the next wall has a large jump outwards. Behavior is there in both 2.4.58.1 and 2.4.58.2 nightly:

[image]

### Project file & How to reproduce

[OP10-Fix.3mf.txt](https://github.com/supermerill/SuperSlicer/files/8337003/OP10-Fix.3mf.txt)

### Version

2.4.58.1

CCS86, 2022-03-23:
Found another one:

[image]

liftbag, 2022-03-24:
I've found a similar issue with spiral vase, but it appears to be a PrusaSlicer 2.4.1 issue. Can you check with PrusaSlicer too?

supermerill, 2022-03-24:
can't reproduce.
it's random?

for the "prusaslicer2.4.1 issue", i'm not sure. If it is a perimeter issue, it can't comes from prusaslicer, as i don't use their perimeter code. But if it's a gcode-viewer issue, then maybe.

[image]

closed 2022-07-14 by supermerill