supermerill/SuperSlicer #2570: Massive overextrusion

robby28-11, 2022-03-24:
### What happened?

When using the 2.4.58.1 I get a massive overextrusion. But I think only the infill is the issue, since the perimeters are looking ok.
A tolerance test failed massively. Same stl with same config and using 2.3.57.11 comes out perfectly - all tolerance pieces can be moved easily. With 2.4.58.1 not a single one is moveable and the surface looks very rough.

Even if I can't see an issue with the gcode, I bet there is a misconfigured setting... any suggestions?

Many thanks in advance

### Project file & How to reproduce

Project exported with 2.4.58.1.
[Tolerance Coin 0530.zip](https://github.com/supermerill/SuperSlicer/files/8339660/Tolerance.Coin.0530.zip)

### Version

2.4.58.1

### Operating system

Windows 11 Pro

### Printer model

Prusa Bear Mk2.5s MMU2s

supermerill, 2022-03-24:
There is a 0.07% deviation from 57.11 to 58.2 for the filament used. These may comes from a slightly different bridge position, so it shouldn't have any effect.

How can you get 'massive overetrusion'?

robby28-11, 2022-03-24:
Yeah, that can't be the issue. Maybe it's not "overextrusion"?
Mabye you can see something more from the pictures. Black one is from 57.11 white one I stopped in the middle of the print with 58.2. (I tried it with the same black filament already with 58.0 - same issue).
As you can see, the first few layers are already looking weird. Infill looks overextruded to me?
All the tolerance test pieces are bonded together.

I have absolutely no idea what is happening here... switching back to 57.11 solves the problem instantly. So I think this is not printer related.

[image]
[image]

robby28-11, 2022-03-24:
Nope, as said before. Also with black and 2.4.58 it is looking weird and all pieces are bonded together. Just can't find the black piece.
White with 2.3.57 is perfect.

robby28-11, 2022-03-24:
Just checked slicing results:
2.3.57: 3,82m
2.4.58: 3,54m

So 2.4.58 is using even 7% less filament?!

This all does not make sense to me. Something weird going on.

supermerill, 2022-03-24:
the 2.4.58 correctly read the start_gcode (and so the purge line) contrary to 2.3, so be sure to remove all custom gcode before comparing.

supermerill, 2022-03-24:
can you:
 - slice with 2.3.57, save the project & the gcode file.
 - slice with 2.4.58 (by opening the project just sved), slice, save the project file in a different file, and the gcode file.
 - give me these 4 file in a zip here?

robby28-11, 2022-03-24:
Thanks for your support. Please find the files in the attachment.

[Project_and_gcode.zip](https://github.com/supermerill/SuperSlicer/files/8341483/Project_and_gcode.zip)

neophyl, 2022-03-24:
For a start check your slice.  The tolerance tests for the closest tolerance (05) commonly are sliced wrong in PS and derivatives  due to the Slice Gap closing radius setting.  Its set to 0.049 by default which results in it being sliced solid in that area.  So its not surprising that part of it is printing solid.  Its sliced that way.  I have all my profiles on both slicers set to add an extra zero in there so 0.005.
Saw this 'issue' multiple times on the Prusa forum with people having problems with these tolerance tests.  

What is surprising is if 2.3 is not doing the same.  Unless the value happens to be set differently in your projects.  Unfortunately I cant open your project in a SuSi2.3 version.

robby28-11, 2022-03-24:
Yeah, I linked the wrong part in the initial post. I'm also having troubles with the 1035 tolerance test in 58.2.
Tbh, all files printed with 58.2 look very rough and "ugly". Already the first layer is very rough in the "infill" section. Perimeters are looking good.

neophyl, 2022-03-24:
The tolerance test included in the later zip is different to the one in the original zip and doesnt have as close a gap so slices fine in both 57 and 58 with the project setting for slice gap closing radius.

Are you 100%  certain that you used THIS model on your massively over extruded print or did you happen to use the one with the 05 clearance for that one ?

robby28-11, 2022-03-24:
Yes, 100% sure.

neophyl, 2022-03-24:
The reason I asked which one is the white one pictured is not sliced with a gap , the infill goes back and forth over the area which it wouldn't if it was following perimeters and were stuck together by over extrusion.  That would suggest that its the slice gap closing radius at play.

robby28-11, 2022-03-24:
Yep, but even on the white, I could hardly remove one of the pieces, all others are already bonded to the main part.

robby28-11, 2022-03-25:
Just reprinted both gcode files from above. Somehow now the issue did not appear. 
Sorry for the confusing, must have been a local problem. I'll keep you updated if it ever happens again.

closed 2022-03-25 by robby28-11