supermerill/SuperSlicer #2572: Extrusion multiplier per feature type e.g., solid infill

rsurgiewicz, 2022-03-24:
I'd like to ask for the feature to have a possibility to set extrusion multiplier per feature type.

E.g., to be able to set different extrusion multiplier for solid infill only.
I am not talking about *extrusion width* but a multiplier. 

So in the end user can set that solid infill will be printed with higher/lower flow that the rest of the print.

If the feature is already present please help me finding it :)

Regards

AndehX, 2022-06-15:
I'd like to respectfully disagree and formally request this feature too.  I frequently have the issue of solid infills having the "pillowing" effect where it looks like over extrusion.  With the ability to reduce the flow rate during solid infills, this issue would be fixed.

Kaelum, 2023-05-06:
This is actually a bug, and is over extrusion.  I just verified this on 2 different printers, 1) Rat Rig V-Core 3.1 and 2) Ender 5 Plus.  With the 2.4.58.5 release of SuperSlicer, it is now over extruding by 25%.  If I drop the extrusion rate to 80% in Klipper, everything prints absolutely perfectly.  This is a very serious issue, and needs to be addressed quickly.  Until then, I'm forced to go back to PrusaSlicer.