supermerill/SuperSlicer #2573: Cannot open file due to error

CD1010, 2022-03-24:
### Version
_Version of SuperSlicer used goes here (help->about)_
2.3.57

### Operating system type + version
Windows 10 

### Behavior
* _Describe the problem_
Cannot open file, i get this error: Failed loading file "Nautilus ENDER BODY.3mf" due to an invalid configuration.
Invalid value provided for parameter support_material_interface_pattern: 

* _Steps needed to reproduce the problem_
Load file that was authored and saved in superslicer.  No error at time of save
* _Expected Results_
[Nautilus ENDER BODY.zip](https://github.com/supermerill/SuperSlicer/files/8343774/Nautilus.ENDER.BODY.zip)

Always able to open file.  If there is an error, code should try to set to some default, warn user but open the file.

* _Actual Results_
File is unopenable and all work lost
  * _Screenshots if useful_

_Is this a new feature request?_
Related guides for writing feature requests: http://meta.stackexchange.com/a/259196 http://nickohrn.com/2013/09/write-great-feature-request-bug-report/

#### Project File (.3MF) where problem occurs
_*Upload a Project File* (.3MF) `Plater -> Save Project as...`_
_You have to zip it (or rename it to .zip) to be able to drag& drop it here_

supermerill, 2022-03-24:
same issue as #1983 #997
the pattern don't save as expected and i don't know why.

[2573_Nautilus ENDER BODY.zip](https://github.com/supermerill/SuperSlicer/files/8344368/2573_Nautilus.ENDER.BODY.zip)

CD1010, 2022-03-25:
You should fix by filling in missing data with default and warning user.  Then figure out why it did not save in first place.
Could have been prusaslicer interference but i am not sure.

supermerill, 2022-03-25:
> You should fix by filling in missing data with default and warning user.

that's what I have done in the 2.4

closed 2022-08-05 by CD1010