supermerill/SuperSlicer #2574: Short Layer time cooling approximate seconds

arne182, 2022-03-24:
**Is your feature request related to a problem? Please describe.**
Short Layer time cooling approximate seconds is an integer:
[image]

**Describe the solution you'd like**
For me I require it to be about 0.4s but as it is an integer I can only set 1s which does not work for my setup.

**Describe how it would work**
same implementation just as float instead of integer 
This was like this in Version 2.3.57.11

supermerill, 2022-03-24:
why?

arne182, 2022-03-24:
[image]
500 x500 x500
The travel speed can easily do 1000mm/s and 2000mm/s with the new 48V drivers

closed 2022-07-14 by supermerill