supermerill/SuperSlicer #2578: Wrong Filament TAG in Export Name

Durahl, 2022-03-25:
### What happened?

My Voron 2.4 is equipped with an [ERCF](https://github.com/EtteGit/EnragedRabbitProject) MMU unit using a single Nozzle.

I noticed a Bug - I assume to have migrated over from PS as its present there too - where the `Filament_Type` in the `Output File > Output filename format` ( mine's: _**[filament_type]** [input_filename_base] [total_weight]g [print_time].gcode_ ) appears to be using the 1st Filament regardless of if it is being used or not.

For example, I had to 3D print an **ABS** Filament Card using the 5th Filament Slot ( Tool=4 )with most of the other slots - including the first one ( Tool=0 ) - being populated with **ASA** ones.

While the Slicers are luckily applying the critically correct settings related to the used ABS Filament the Output File would "revert" back to using the 1st Filaments Material Type.

### Project file & How to reproduce

GCODE File downloaded from Printer with ASA Filament Type in name despite Tool=4 being used which is being declared as ABS in the file ( at the end of GCode ) [ASA Filament_Card 3.29443g 13m.zip](https://github.com/supermerill/SuperSlicer/files/8354087/ASA.Filament_Card.3.29443g.13m.zip)

### Version

2.4.58

### Operating system

Windows 11

### Printer model

Voron 2.4 with StealthBurner & ERCF

supermerill, 2022-03-26:
for the time being, you can use `{filament_type[initial_extruder]}`

Durahl, 2022-03-26:
Sweet! That worked splendidly!
Thanks again! 😁

closed 2022-03-26 by Durahl