supermerill/SuperSlicer #2581: Nightly build incorrect

dingyifei, 2022-03-26:
### What happened?

The nightly build for Ubuntu-GTK3/Ubuntu is 2.3.57, the wrong branch is probably used 
[image]

### Project file & How to reproduce

Download and run the AppImages, they are all the wrong version

### Version

2.3.57

### Operating system

Fedora release 35

supermerill, 2022-03-28:
btw, i'm moving to nightly_dev for the dev branch (currently 2.4) and nightly_master for the master branch (currently 2.3)

dingyifei, 2022-03-28:
Additionally I just realized the release candidate for linux doesn't have the UI improvements

supermerill, 2022-03-28:
oh, i see why. my bad.

dingyifei, 2022-03-28:
> oh, i see why. my bad.

am I using an old version the whole time lmao

supermerill, 2022-03-29:
the build is right, my commit message also, but there was a bug in the build script, so it uses the old code to build instead of the one from the commit.

dingyifei, 2022-03-30:
It works now

closed 2022-03-30 by dingyifei