supermerill/SuperSlicer #2582: Universal Settings ala Cura

inspectionsbybob, 2022-03-27:
Cura plugin introduced a new feature called "universal settings" which does multiple tweaks depending on the users intent.

They currently have options for things like:
Bed Adhesion
Small figures
top surface
support

My suggestion is to have a set of sliders that determine the importance of certain aspects.

having sliders for:
Speed
Surface finish
Adhesion
Supports
Overhangs

so that the user can select what is important to them for this print... and then internally set the parameter to favor whatever they have chosen.

supermerill, 2022-03-27:
you can create some quick settings that does that, with the scripted widget.
take example on the `s_support_fff`, defined in `resources\ui_layout\example\freq_fff.ui` and `resources\ui_layout\example\print.as`
some info about functions available in script are in the print.as file.
info about the .UI syntax are in the readme

inspectionsbybob, 2022-03-27:
I tried clicking on the links and got a 404 error.

Bob S

supermerill, 2022-03-28:
?
there are no link

inspectionsbybob, 2022-03-28:
Ah, when viewed in email the "support_fff" looked like a hyperlink to a document.