supermerill/SuperSlicer #2583: Brim overwrites Skirt

inspectionsbybob, 2022-03-28:
2.3.57
_Version of SuperSlicer used goes here (help->about)_

Win10 with all patches
_What OS are you using, and state any version #s_

I went to print a difficult print and enabled both Skirt AND Brim

It laid out the Skirt and then proceeded to Put the Brim OVER portions of the Skirt. The Brim didn't care the skirt was there
so there was no real problem, but it was an unexpected behavior.
In other slicers the Skirt is OUTSIDE the Brim(s)

_Is this a new feature request?_
No

supermerill, 2022-03-28:
did you have the "skirt distance from brim" selected?

inspectionsbybob, 2022-03-28:
Ah...No...
Coming from Cura I didn't know about that one...
I will try that later today,  likely will fix it...

PS... I tried SS over a year ago, this new version is very very good...
much better than it was before.

-- 
-----------------------------------------------------------------------------------
Need a quick answer? Call our booking line at (301) 208-8289, or book your
inspection on our website 24/7.
Don't wait for a response as someone else may get the spot you want... call
Today!
-----------------------------------------------------------------------------------
Bob Sisson, ACI, BVI
Inspections by Bob, LLC
www.inspectionsbybob.com
***@***.***
MD Lic#29666; ASHI Member #212016

supermerill, 2022-03-28:
It's checked by default, so someone unchecked it manually.

inspectionsbybob, 2022-03-28:
I just tried it after making sure that it was enabled, and it did the same thing again

[image]

inspectionsbybob, 2022-10-03:
It is the setting for distance from...

closed 2022-10-03 by inspectionsbybob

inspectionsbybob, 2022-10-11:
> [image: image]
> <https://user-images.githubusercontent.com/6536403/160432372-405ead15-2cb7-41d0-8a77-c52708236bc7.png>
>
> how can I emphasis the "include your project file" in the github issue
> form? You're the thrid in a row to not include it and i'm a 'bit' grumpy
> about that now.

Sorry, the project file is big and messy... I will try to reproduce it with
a smaller project.
I think it has something to do with multiple objects, but I will see what I
can reproduce on a smaller scale.