supermerill/SuperSlicer #2584: Custom G-code for temperatur setting

joluxer, 2022-03-28:
**Is your feature request related to a problem? Please describe.**
Using the proportional auto-temperature feature of Marlin, I cannot find a consistent way to insert the temperature change from first layer to other layers. After layer 0 is always M104 and M140 inserted, regardless of those settings, I can have in before/after layer change code.

**Describe the solution you'd like**
I see two possible solutions:
* slicer detects the use of M104 and/or M140 etc. in layer/tool/feature change code, simelar to its detection for start g-code, leaving out its own settings
* custom G-code section for temperature settings along with other custon G code sections, but this seems inconsistent, as the temperature change for features/tools/layers could be different

**Describe how it would work**
Using the same macro language set as it is available for start G code and others.

**Describe alternatives you've considered**
I tried to insert temperature settings using M104/M140 in before layer change and also in after layer change code, both did not work, as the emitted G code still contains plain temperature settings after the change to layer 1.
Post-processing the G code in scripts is hard, as there is no easy identification of the temperature change on feature/tool/layer change

supermerill, 2022-03-28:
i'm more leaning toward detecting temp change from one of the two layer change gcode.

joluxer, 2022-03-28:
So to keep it simple for users, i would kindly suggest to inspect both layer change codes by Superslicer so noone needs to keep in mind, which G-code set is the right one, if this is not too much work.

By the way, I never inspected the capabilities of the other custom G-code sets (tool change, object change, etc.), what happens, if temperature setting or other things, like machine limits etc., are preconfigured. Perhaps it might be easier to just add an checkbox simelar to `start_gcode_manual` to every G-code set and blend everthing and handing full responsibility over to the custom G-code set. Just a few pence of ideas again.

supermerill, 2022-03-28:
I'll add scritpable gcode writer later this year, to be able to change how the gcode writer write the file (to allow output to not-gcode firmware). If someone want something really custom, he will be able to change how it works that way.