supermerill/SuperSlicer #2585: Support Enforcer Doesn't

inspectionsbybob, 2022-03-28:
### Version
2.3.57

### Operating system type + version
Win10

### Behavior
Support Enforcer doesn't create supports when used with auto supports.
I have a Model that is tall and skinny and it has failed every time when it gets "too" tall and starts to wiggle, even with Raft or Brim.
I placed a "Support Enforcer" on the part just below the level it fails at but it did not generate anything at that location.
Not sure if is BUG or deliberate, but I need to know how to put a support where SE doesn't auto generate one.

Here is where support is needed
[image]

And nothing was applied
[image]

_Is this a new feature request?_
No.

#### Project File (.3MF) where problem occurs
_*Upload a Project File* (.3MF) `Plater -> Save Project as...`_
_You have to zip it (or rename it to .zip) to be able to drag& drop it here_

supermerill, 2022-03-28:
I tested quickly with some cube, it can auto support + enforcers.

inspectionsbybob, 2022-03-28:
I tried to Zip the 3Mf file, but it won't let me attach the Zip file.

inspectionsbybob, 2022-03-28:
Lets try it this way....
Here is a link to a Google download of the 3MF file

https://drive.google.com/file/d/1tNYOxnC0L-46gHS1klJcPqG746KkxLL6/view?usp=sharing

supermerill, 2022-03-29:
the leg angle is very sharp. With the support separation used, there is nothing left to support.

[tinkerbell_lower body.zip](https://github.com/supermerill/SuperSlicer/files/8372013/tinkerbell_lower.body.zip)

inspectionsbybob, 2022-03-29:
This model is representative of the problem I have with tall
skinny things.  Cura has a plug-in support that offers "Butress"
supports... I guess for some models I will still have to use Cura...
Please don't wait for me to respond when booking an inspection. I turn off
my devices when on appointments and someone else could get your spot.
Please CALL our service or book yourself online to save the spot as soon as
you know the time you want.

Bob Sisson ACI, BVI
Inspections by Bob
ASHI Member # 212016
MD Home inspectors License #29666*
*MAC-ASHI Chapter President 2010-2012
ASHI Mid-Atlantic Group Leader 2009-2012
ASHI National Board Of Directors 2012-2015, 2017,2018-2020
***@***.***
www.inspectionsbybob.com
(301) 208-8289 Scheduling/Pricing

inspectionsbybob, 2022-10-03:
New versions are doing better

closed 2022-10-03 by inspectionsbybob