supermerill/SuperSlicer #2586: Perimeter Overlap is not controlling spacing

mirasderbissalin, 2022-03-28:
### Version
2.3.57.11

### Operating system type + version
Windows 10

### Behavior
* _Describe the problem_
Overlap doesn't control spacing. When setting overlat to different from 100% spacing doesn't change.

* _Steps needed to reproduce the problem_
Create CR10 profile, change nozzle to 0.6,  width ext perimeter 0.65, support 0.4, rest 0.7, vary overlap.

* _Expected Results_
Overlap Default supposed to change spacing, it doesn't.
When printing with all perimeters (I set like 30 to be sure) model is supposed to be 100% with no overextrusion.

* _Actual Results_
Spacing for every other 0.7 extrusion is 0.6149, but for internal perimeters is 0.5699.
Perimeters are overextruded and model bulges on the top

  * _Screenshots if useful_
[image]
[image]
[image]
I tried to look into other profiles, and Voron 2.4 volcano 0.6 has much more appropriate spacing when set to 0.7 width
[image]

_Is this a new feature request?_
No

#### Project File (.3MF) where problem occurs
That happens always

supermerill, 2022-03-28:
[image]

supermerill, 2022-03-28:
oh, i understand the issue.
it's a refresh problem that have been fixed in 2.4 but not in 2.3

mirasderbissalin, 2022-03-28:
Oh, thanks, good to know!

> a gui issue. the compute behind is good.

So... my overextruding perimeters and underextruding tops problems are somewhere else? F@c<, I thought I found it

supermerill, 2022-03-29:
You don't need to touch overlaps settings until everything is perfect with basic pla.

adamrumjahn, 2022-04-26:
Also, it seems that there is no way to change the infill spacing via overlap, is this correct? is this always a fixed value with no changes permitted in overlap?

adamrumjahn, 2022-04-26:
When I examine the Gcode, it looks like that either the extrusion perimeter spacing or the extrusion perimeter width is blank, depending on which one I updated last.  Is this behaviour normal?
[image]

[image]

supermerill, 2022-04-26:
> When I examine the Gcode, it looks like that either the extrusion perimeter spacing or the extrusion perimeter width is blank, depending on which one I updated last. Is this behaviour normal?

yes

> I'm using the latest nightly build : Version 2.4.58.2 and it still either seems to be failing to update, or not calculating properly.

latest nightly is 2.4.58.3, and can't reproduce your behavior.

lfrlucas, 2022-08-03:
Not sure if related, but how can I change spacing and width values for infill to be both 100%? I'm not finding any overlay setting for infill.

supermerill, 2022-08-04:
> Not sure if related, but how can I change spacing and width values for infill to be both 100%? I'm not finding any overlay setting for infill.

not possible. look at the tooltip, it should say you that only one of them can be the "active" one. The othe is just a computed value from the active.

It's like asking to set the number of cups of flour and the weight of the floor to incompatible values.

lfrlucas, 2022-08-04:
@supermerill but the same is not true e.g. for "perimeter" widht and spacing.
[image]
I can change the perimeter overlap here:
[image]
and thus obtain different relations between perimeter width and spacing.
Looks like the "infill overlap" is fixed and not configurable like "perimeter overlap".

supermerill, 2022-08-04:
if you want a global under-extrusion, reduce the max filament overlap.

lfrlucas, 2022-08-04:
Ahh I didn't know about that setting. This means we have some control on it. Thanks!

supermerill, 2022-08-05:
but I still don't understand. Infill lines doesn't touch each other, why do you want to touch the overlap?

lfrlucas, 2022-08-05:
I was trying to setup SS similarly to my Cura profile, which uses 0.4 width for all line types (and I though spacing in cura was the same but maybe I'm wrong). At least for solid infill, after a few tests, it seems that my Ender 3 V2 gets better results in top surface using solid infill width = spacing. I could make it by setting "solid fill overlap" to 0%. Note that I have E-steps calibrated. I just wanted to try also the same for other line types, like internal infill, but I gave up ( max filament overlap affects all line types, and at least for the first layer I still need 100% overlap).

supermerill, 2022-08-05:
For top surface, there is the solid infill overlap.
But I discourage to use it, unless you already tried everything else (flow, temperature, speed, extruder).

lfrlucas, 2022-08-07:
Looks like I went to play with solid infill overlap too early. As you said it seems I can improve the results using other settings and keeping default solid infill overlap.