supermerill/SuperSlicer #2587: NAME objects other than "Object 0"

inspectionsbybob, 2022-03-28:
**Is your feature request related to a problem? Please describe.**
When printing Multiple Objects what appears on the Printer is:
"Printing Object 0"
"Printing Object 1"
"Printing Object 2"
and so on...

If I want to cancel an object because of an error or failure but let the rest of the print continue
I would love to have the NAME from the Platter screen be used so I could cancel "Wing #4" rather than wondering which object is which...

**Describe the solution you'd like**
Have the Object Name(s) be picked up from the List on the Platter Page

**Describe how it would work**
Have the Object Name(s) be picked up from the List on the Platter Page
-AND/OR- be able to edit the name(s) on the Platter screen so they FIT into the 12 character printer field

**Describe alternatives you've considered**
None

**Additional context**
[image]

so it appears on this screen

[image]

supermerill, 2022-09-07:
can you send me a project & the gcode file of an example that produce that?

inspectionsbybob, 2022-09-07:
The Current Version 2.4.58.4 behaves differently as it no longer includes any M117's of "Printing Object 4" or anything like that.

I am now using something Like Octoprint (Actually "Repeatier-Server" but not everyone knows about that) and it does something different that lets me cancel an object.

THIS ticket can be closed as "fixed" but I may open something similar about no M117 messages when moving between objects.

closed 2022-09-07 by inspectionsbybob