supermerill/SuperSlicer #2589: Pb with vase mode

Nossolov, 2022-03-28:
### What happened?

For this part : https://www.thingiverse.com/thing:2537701
Pb at 19.5mm height.
Try to repair object without effect.
Try with equivalent version of PrusaSlicer : everything is Ok.

### Project file & How to reproduce

https://www.thingiverse.com/thing:2537701

### Version

2.3.56

### Operating system

Win10

### Printer model

Anet A8

Nossolov, 2022-03-31:
Bonjour.
Here is my project file. I just change extention name from .3mf to .txt for uploading.

Superslicer is a very nice thing. Thank you !

Bonne fin de journée.

[spiral_vase-linear_bearing.txt](https://github.com/supermerill/SuperSlicer/files/8387412/spiral_vase-linear_bearing.txt)

supermerill, 2022-03-31:
something in the stl isn't sliced well. Doesn't appear if re-meshed by meshmixer, or if the slices are in different height, or by using the 2.4 new slicing code.

Nossolov, 2022-03-31:
Ok, thank you.
I wait for 2.4 version, may be with snug support...

closed 2022-03-31 by Nossolov