supermerill/SuperSlicer #2590: 2.4.58.1 (Pre-release) Crash on startup

Juusoz, 2022-03-29:
### What happened?

The new Superslicer version crashes on startup on my workplace's (fairly old) computer.

### Project file & How to reproduce

Booting it up with the console mode gives this error:

[image]

Seems it can't open the **_print.as_** file in %appdata%. It opens just fine with notepad++, so it doesn't seem like it is corrupted either.

### Version

2.4.58.1

### Operating system

Windows 10 Home 21H2

supermerill, 2022-03-29:
Just to be sure, you already launched an older version before, as requested by the "known issues" of the release page?

Juusoz, 2022-03-30:
Yeah, I already had an older version on this computer before.

One thing to note is that if I delete the Appdata\Roaming\Superslicer folder, the newer version can't make a new one, but the older one can.

Edit: Got it working. I deleted all of the files in Appdata\Roaming\Superslicer, but not the folder itself.
Just remember to backup the config files using a previous version before you delete them.

I do actually have the old Appdata\Roaming\Superslicer folder backed up, it might be helpful in understanding what went wrong.
[SuperSlicer.zip](https://github.com/supermerill/SuperSlicer/files/8378947/SuperSlicer.zip)

Juusoz, 2022-03-30:
Correction, got it working **once**. Trying to open it a second time will result in the original error:

[image]

It shouldn't be caused by any profiles at least, considering I hadn't imported them back yet.

supermerill, 2022-03-30:
can you try the latest nightly? https://github.com/supermerill/SuperSlicer/suites/5835274444/artifacts/196041452

Mosher23, 2022-03-30:
Same here on Mac OS 12.3. Crashes on start-up

Juusoz, 2022-03-31:
> can you try the latest nightly? https://github.com/supermerill/SuperSlicer/suites/5835274444/artifacts/196041452

Same error, except now deleting the Appdata files also doesn't do anything.

[image]