supermerill/SuperSlicer #2591: No infill on bottom layer of part with extremely thin walls

LeavOA, 2022-03-29:
### What happened?

This may be a "working as intended"

But Basically you can see that at 100% scale, the bottom layer does not have any infill.
Scaling to 200% allows SS to generate infill.

Not sure what parameters I can change to fix the 100% version

[image]

### Project file & How to reproduce

[Fidget_Switch_STL.zip](https://github.com/supermerill/SuperSlicer/files/8370998/Fidget_Switch_STL.zip)

### Version

Version 2.3.57.11

### Operating system

Windows 8.1

### Printer model

Custom based on Duet 2 wifi

tome9111991, 2022-03-29:
check your "Anchor solid infill by X mm"
with default value i have infill

supermerill, 2022-03-29:
It's a bug but i won't fix it right now, as it's a very edge case.
I guess there is a surface that is enlarged so much, it began to overlap over everything else.