supermerill/SuperSlicer #2594: Appimage icon

dingyifei, 2022-03-29:
**Is your feature request related to a problem? Please describe.**
The appimage doesn't contain a default icon. The icon only shows up when the program is running.  

**Describe the solution you'd like**
Default icon for appimage

**Describe how it would work**
Basically copy the icon to another location inside the appimage. Some prusa slicer users came up with a solution but probably gave up on it for some reason...
https://github.com/prusa3d/PrusaSlicer/issues/7801

**Describe alternatives you've considered**
no icon is fine

**Additional context**
Add any other context or paint drawing about the feature request here.

supermerill, 2022-03-30:
There is already a SuperSlicer.png at the  appimage root.
Why it doesn't work on your linux?

dingyifei, 2022-03-30:
It seems to be a common issue, where the icon isn't in usr/share/icons of the appimage.

https://github.com/AppImage/libappimage/issues/165

I'm running fedora 35 with the default Gnome installation, where the LycheeSlicer AppImage has an icon but SS has no icon. It is strange when I extract and re-pack the LyCheeSlicer, the new AppImage doesn't have an icon...

bassamanator, 2022-04-02:
I'm on Manjaro, using today's release. No icon is showing for me.

| SuperSlicer Version 2.3.57 SuperSlicer-ubuntu_18.04-gtk2-2.3.57.12.AppImage
| Operating System: Manjaro Linux
KDE Plasma Version: 5.24.3
KDE Frameworks Version: 5.91.0
Qt Version: 5.15.3
Kernel Version: 5.15.28-1-MANJARO (64-bit)
Graphics Platform: X11
Processors: 16 × AMD Ryzen 7 2700X Eight-Core Processor
Memory: 31.4 GiB of RAM
Graphics Processor: AMD Radeon RX 580 Series

supermerill, 2022-04-02:
what i have to do for it to appear?

dingyifei, 2022-04-03:
> what i have to do for it to appear?

based on this issue, https://github.com/AppImage/libappimage/issues/165, add a folder in appimage with the same icon.

supermerill, 2022-04-03:
so, with `/` the root directory inside the appimage archive, I have to copy my `/SuperSlicer.png` to `/usr/share/icons/hicolor/192x192/apps/SuperSlicer.png` ?

dingyifei, 2022-04-03:
> so, with `/` the root directory inside the appimage archive, I have to copy my (current) `/SuperSlicer.png` to `/usr/share/icons/hicolor/192x192/apps/SuperSlicer.png` ?

I suspect that would fix it, when I was comparing the lycheeslicer appimage with superslicer's, the .desktop is similar.

bassamanator, 2022-04-04:
@supermerill Post a test-fix here and I'll try it out and report back the results.

supermerill, 2022-04-05:
@bassamanator 
https://github.com/supermerill/SuperSlicer/actions/runs/2091250726 or the gtk2 nightly if it's better

bassamanator, 2022-04-06:
@supermerill I tried SuperSlicer-gtk3.AppImage. Icon is now showing on my system.

dingyifei, 2022-04-16:
Seems to be partially solved, icon appear correctly now
but when it starts, it shows a separate thing in the taskbar

[image]

bassamanator, 2022-04-17:
> Seems to be partially solved, icon appear correctly now but when it starts, it shows a separate thing in the taskbar

Does this separate thing disappear? That separate thing is likely an hourglass animation.

dingyifei, 2022-04-17:
> > Seems to be partially solved, icon appear correctly now but when it starts, it shows a separate thing in the taskbar
> 
> 
> 
> Does this separate thing disappear? That separate thing is likely an hourglass animation.
> 
> 
> 
> 
> 
> 

It disappear when it's closed, but not when it's running

bassamanator, 2022-04-18:
> It disappear when it's closed, but not when it's running

I see. I can't imagine this having anything to do with SuperSlicer's appimage icon in particular.

dingyifei, 2022-04-18:
> > It disappear when it's closed, but not when it's running
> 
> 
> 
> I see. I can't imagine this having anything to do with SuperSlicer's appimage icon in particular.
> 
> 

I think it's an issue related to superslicer's appimage setup, maybe I should open another issue?

bassamanator, 2022-04-18:
@dingyifei I'd be curious to see how other Appimages are working on your computer. Could you maybe assign the SS icon to a different appimage on  your machine and see what happens when you start the app?

supermerill, 2024-05-21:
i'll close because i can't reproduce. If it's disapearing agin, fell free to reopen it.

closed 2024-05-21 by supermerill