supermerill/SuperSlicer #2598: Only infill where needed not working with Supporting dense layer

markcarwinbarr, 2022-03-30:
Using the 2.3.57.11 and 2.4.58.1 versions, checking "Supporting dense layer" under the Infill settings disallows using "Only infill where needed". This happens to any file I've tried to slice on Windows 10.

In previous versions that I've used (2.3.56.6 and 2.3.57.2), these two options work fine when both are checked so I'm assuming this is unexpected behavior. Also, I can have both options checked by clicking "Only infill where needed" before "Supporting dense layer", but doing this still greys out "Only infill where needed" and doesn't seem to activate it's functionality. It would be great to be able to use both of these options together along with the newer version updates. 

The images below are taken from version 2.3.57.11

[image]

supermerill, 2022-03-30:
It currently doesn't work nice together
#1657

The suppporting dense layer currently can't handle void layers created by the "combine infill" (or it's just luck) 
I have to create a new algo that find a way to merge sparse & voids and "recover" from that merge.

markcarwinbarr, 2022-03-30:
Ah ok, thanks for taking a look and I'll keep an eye out for that in future updates.