supermerill/SuperSlicer #2599: SET_PRESSURE_ADVANCE doesn't specify EXTRUDER parameter, produces error in Klipper

justynikeda, 2022-03-30:
### What happened?

I'm running a printer with a 2-in-1-out Y adapter on the hotend. Klipper requires an extruder parameter to be specified when using SET_PRESSURE_ADVANCE with multiple extruders. When the EXTRUDER parameter is missing, Klipper throws an error and cancels the print.

I have all of my pressure advance and toolchange commands handled in macros, but when using the "Klipper" flavor of firmware in Superslicer's printer settings, Superslicer inserts a "SET_PRESSURE_ADVANCE ADVANCE=0" before the toolchange, even though I have my own custom toolchange g-code set.

Switching the g-code flavor to Marlin removes the SET_PRESSURE_ADVANCE command from the produced g-code and the file is then printable.

Perhaps removing the automatic insertion of SET_PRESSURE_ADVANCE commands when custom toolchange g-code is present would be best. I had written my toolchange g-code under the assumption that this was the case anyway  (my toolchange macro disables pressure advance at the start, and re-enables it after). The tooltip would have to be updated to ensure the user is aware of that.

### Project file & How to reproduce

[pressureadvance_example.zip](https://github.com/supermerill/SuperSlicer/files/8376168/pressureadvance_example.zip)
[pressureadvance_examplegcode.zip](https://github.com/supermerill/SuperSlicer/files/8376173/pressureadvance_examplegcode.zip)

The project file and the produced gcode are included. If you inspect the gcode and search for "SET_PRESSURE_ADVANCE", you'll find one before the toolchange.

### Version

Version 2.3.57

### Printer model

Elegoo Neptune 2 with Y-adapter dual extruder

supermerill, 2022-03-30:
that's how the wipe tower works: it's a big blog on its sied that do its thing alone.
I want to get rid of this an dintegrate it properly into the slic3r, but
 * that will make merging wipetower thigns from prusa harder
 * it's a lot of work.

for your issue, in the klipper cod, it says `If EXTRUDER is not specified, it defaults to the stepper defined in the active hotend.` so I don't know why it complains.
https://www.klipper3d.org/G-Codes.html?h=set_pressure_advance#set_pressure_advance

The way to set the pressure advance problem is to init it in the filament start gcode.
It has to be deactivated for the purge/ramming. It will be reactivated at the filament loading, just after.

justynikeda, 2022-03-30:
> for your issue, in the klipper cod, it says If EXTRUDER is not specified, it defaults to the stepper defined in the active hotend.

Yeah, it used to / supposed to do that, but they deprecated the shared_heater function and now require a SYNC_EXTRUDER_MOTION command. Setting that up for a 2-in-1-out requires me to disable the main extruder stepper, so when klipper looks for the default "extruder", it's disabled and errors out. But this is more an issue on Klipper's end.

Well, I understand the necessity of it. I guess for my setup I will have to use workarounds until Klipper can resolve the error. For now I will use the Marlin flavor gcode as that seems to work fine. Any real 'fix' would require drastic changes to the wipe tower and I understand that's asking quite a lot!

Normally I would write a gcode override macro to replace 'SET_PRESSURE_ADVANCE' with 'SET_PRESSURE_ADVANCE EXTRUDER={whatever}' but as far as I'm aware there is no parameter to reference the current motion sync extruder.

Thanks anyway!

closed 2022-03-30 by justynikeda

supermerill, 2022-03-30:
I guess you can add a `SET_CURR_EXTRUDER  S{current_extruder}` in your filament start gcode, and use your SET_CURR_EXTRUDER  klipper macro to store it into a klipper variable.

cpfarhood, 2022-07-08:
It's fine to disable pressure advance for ramming, but it's broken without specifying an extruder.