supermerill/SuperSlicer #2601: Equivalent for "Slicing Tolerance" in Cura

logiclrd, 2022-03-30:
**Describe the solution you'd like**
It is entirely possible that SuperSlicer already has equivalent functionality, and I just don't know where to find it. :-)

I recently saw this video: https://www.youtube.com/watch?v=X6nCKQToOUg&t=1s

It describes a setting in Cura that lets you control whether each layer is sliced based on whether, at each point, any Z intersects the part, or any Z _doesn't_ intersect the part, or an in-between of the two.

[image]

[image]

**Describe how it would work**
I would like the ability to indicate how each slice's path should be derived from the slice of the model with at least these modes:

* Treat places where any Z value in the slice intersects the model as being _part_ of the model that needs material.
* Treat places where any Z value in the slice _does not_ intersect the model as being _not part_ of the model.
* An intermediate value that is (approximately, if necessary) the average of the two.

**Describe alternatives you've considered**
Without a setting like this, the only alternative is to design the tolerances directly into the model, factoring in which interpretation the slicer is going to use. At the very least, this is likely to require some trial and error, because it may not be easy to find out exactly what interpretation the slicer is using.

supermerill, 2022-03-30:
Nothign is done for something like that 
It's using the polygon at the layer height as the extreme limit for the theoretical extrusion.

something like 
[image]

but if there is an overhangs, it may increase the flow (depending on your settings, and so extrude a bit more plastic.
[image]

supermerill, 2022-03-30:
Crude inefficient ago: you have to cut at the start, end and the slices for all Z for all points that are in-between.
compute the intersection to get the "exclusive"
compute the union to get the "inclusive"

logiclrd, 2022-03-30:
Mm, isn't it possible to just project the slice's geometry orthographically from on top?

logiclrd, 2022-03-30:
I suppose that is perhaps _literally the same thing_ as what you described, when it comes down to it :-P