supermerill/SuperSlicer #2602: Anchor Solid infill by X : Bridged - Not working

PhilBaz, 2022-03-30:
**Bug description**
A clear and concise description of what the bug is.

Anchor Solid infill by X has two parameters default, and bridged.  I would expect bridged would affect only the 'internal bridged infill' layer.  Default parameter appears to control both 'internal bridged infill'  & 'solid infill', bridged has no effect. 

**Expected behavior**
 A clear and concise description of what you expected to happen.

Anchor Solid infill by X : Bridged, this setting/parameter should affect only the 'internal bridged infill' pushing it out to meet the perimeter. Additional solid infill layers should be controlled by the default parameter.

!!>> Project File <<!!
*Please* upload a ZIP archive containing the project file used when the problem arise: `File`->`Export project as...`, then zip it & drop it here.  
Please export it just before the problem occur. Even if you did nothing and/or there is no object, export it! (it contains your current configuration).

[Hoof.zip](https://github.com/supermerill/SuperSlicer/files/8383023/Hoof.zip)

**To Reproduce**
Steps to reproduce the behavior:
1. Go to '...' Infill>advanced> Anchor Solid infill by X : Bridged
2. Click on '....' change setting, nothing happens. 'internal bridged infill' should be pushed to perimeter with higher settings.
3. Scroll down to '....'  Infill>advanced> Anchor Solid infill by X : default
4. See error ...Default controlls both 'internal bridged infill'  & 'solid infill'. T

**Screenshots**
If applicable, add screenshots to help explain your problem.

**Desktop (please complete the following information):**
 - OS: [e.g. MacOS, Linux, Windows] windows
 - Version [e.g. 2.3.55.5], Use `Help->About`_  2.3.57
 - Printer model Voron 2.4

**Additional context**
Add any other context about the problem here.

supermerill, 2022-03-30:
bridged is for the (bottom) bridge, and all solid layer upon it.
default is off the top layer, and all layers below it (including the internal bridge).

I can rename them.

PhilBaz, 2022-03-31:
@supermerill 

Ahh, that's not what I expected. Maybe just clarification in the tool tip. Or if you feel justified renaming.