supermerill/SuperSlicer #2603: Auto Speed not respecting non-zero speeds

Deutherius, 2022-03-30:
### What happened?

Auto Speed tooltip states that it will only adjust max VFR/print speed for extrusion types that have the speed field set to 0. This is not the case.
[image]

Cube sliced with auto speed max VFR at 20 mm^3/s
[image]

Same everything, just auto speed at max VFR of 2 mm^3/s
[image]

Speed tab, only sparse infill has 0 as speed and thus should be the only thing limited by auto speed according to the tooltip
[image]

Same behavior also happens when every speed is set to non-zero (and, thus, nothing should be limited by auto speed).

### Project file & How to reproduce

[auto_speed_test.zip](https://github.com/supermerill/SuperSlicer/files/8384075/auto_speed_test.zip)

### Version

2.4.58.1; also observed independently in 2.3.57.8

### Operating system

-

### Printer model

-

supermerill, 2022-03-31:
Looking at the code, I changed it so the max_volumetric is applied to everything (last july).
I think the rational is that I had many bug report of "I set the max volumetric and it's not applied".
But Why I didn't move it out out of the 'auto-speed' section?

Is it problematic for you that the print max.vol. isn't restricted to auto-speed?
(or do I need to only fix the tooltips & position)

Deutherius, 2022-03-31:
I downloaded 2.3.56.5 to check, and it seems that auto speed had 2 distinct tooltips for VFR and speed:
1) VFR says that it limits volumetric speed of the print to the maximum volumetric feed rate your extruder supports
2) Speed says that if any field is set to 0, it will calculate the optimal speed to keep nozzle pressure constant
This was not working, however, setting any speed to 0 resulted in extremely slow speed for that extrusion type, regardless of auto speed settings.

So if I understand it correctly, Auto Speed has two functions (both working) now:
1) It sets any base speed field with a 0 to max VFR or speed, whichever is lower, and
2) it limits the max VFR and speed of the print (same as max VFR and print speed in the filament tab), regardless of the set speeds

I have noticed some amount of confusion as to the function of auto speed and I think the incorrect (or rather incomplete) tooltip was the cause.

As for me personally, I like the idea of "maxing out" speeds automatically by setting a 0 in the speed field and auto speed handling the rest, it's elegant. I can achieve the same functionality by just setting the speed to 99999 though, so I do not really care if this functionality is kept.

I think adding back the info that it Auto speed max VFR also limits global VFR for all speeds would be sufficient to clarify the confusion.