supermerill/SuperSlicer #2608: I don't even know how to describe this... weird behaviour after 1st layer

alexvonsiebenthal, 2022-03-31:
### What happened?

See video. I had it happen repeatedly, hence I would think it's a slicer problem, not a printer issue. I changed the angle of the model and then it printed fine.

Video: https://youtube.com/shorts/-BqaMG_6GE0?feature=share

### Project file & How to reproduce

[3PotTest v3.3mf.zip](https://github.com/supermerill/SuperSlicer/files/8388190/3PotTest.v3.3mf.zip)

### Version

2.3.57.9

### Operating system

macOS 12.1

### Printer model

voron 2.4 / stealthburner

alexvonsiebenthal, 2022-03-31:
Here's the gcode file to
[3PotTest v3.gcode.zip](https://github.com/supermerill/SuperSlicer/files/8388202/3PotTest.v3.gcode.zip)
o

supermerill, 2022-03-31:
on your gcode, line 2680, you have the move `G1 X215.49 Y151.938`
In your video, you can see that your printer try to execute it but it stalls.
In general, it's because it's hitting something, but as the previous command is `G1 Z1.07 F18000`, it's unlikely.
My guess is that your acceleration/speed is too high and the motor can't keep up. It doesn't write the `FXXX`, as it's using the last value. The last value is `F18000` (300mm/s * 60 to converter into mm/min)
Check that your printer is able to support that speed. There doesn't seems to have any acceleration override, so ensure your setup your firmware with good values.

I can't do anything on the slicer side, I can't sport anything wrong.

alexvonsiebenthal, 2022-03-31:
Thanks for taking the time to look into it!
Strangely, it went away when I rotated the model 45 degrees.
I have limited accelerations in Klipper to 3000/1500 and it can easily handle that.
The 300mm/s are for the travel speed.

alexvonsiebenthal, 2022-03-31:
> Mayeb the 300mm/s is too high for the z axis, and as the z-hop doesn't happen, it hit a little plastic thingy from the first layer?

I don't think so. It's clearly crunching on the A/B motors and even if it was hitting the print, it would just catch and grab it. Those steppers have too much power to be caught by some plastic on a PEI sheet...