supermerill/SuperSlicer #2610: 2.4.58.2 build has file versions for 2.3.57.12

strayr, 2022-04-01:
### What happened?

Look at artifacts on https://github.com/supermerill/SuperSlicer/releases/tag/2.4.58.2
Downloaded zip appears to contain 2.3.57.1
Generates the following errors on run
[image]
[image]
[image]

Help -> about reports 2.3.57.12
[image]
2

### Project file & How to reproduce

download, unzip and run

### Version

2.4.58.2

### Operating system

windows

lordIcocain, 2022-04-01:
If after that 2.4 wont launch delete this stuff
%AppData%\SuperSlicer
[image]

tapir, 2022-04-01:
indeed it looks like the 2.4 release is falsely tagged and names as 2.3

supermerill, 2022-04-01:
I put the asset from the other release i made an hour before ><
fixed

strayr, 2022-04-01:
Awesome, thanks!

closed 2022-04-01 by supermerill