supermerill/SuperSlicer #2611: Toolchange temperature wait for cooldown

PierreMasselot1, 2022-04-01:
**Is your feature request related to a problem? Please describe.**
Currently when setting a tool change temperature the Gcode command (using Marlin Gcode flavor) that is sent is M109 S[temperature]. This command doesn't  wait for the hotend to cool down and going with a really low temperature allows to reduce stringing when performing a tool change (which is the main issue that I had with my multi material upgrade). Marlin supports M109 R[temperature] which waits for the hotend to cool down before moving on. Sending M109 R instead of S will wait for cooldown
Looking at the Reprap Gcode spec this should also be supported on other  flavors but I don't  have a printer to check...

I have already implemented and tested the solution and will be opening a PR that will be linked to this issue.

Thanks for the great work @supermerill  !

supermerill, 2022-04-01:
What is exactly 'setting a tool change temperature the Gcode command' ? There are too many ways of doing something similar.

PierreMasselot1, 2022-04-01:
[image]

It's specifically for single nozzle multi material setups, before extracting the filament during a toolchange, there is some additional logic in order to prevent having massive stringing in the nozzle and bowden tube. Part of that logic is cooling down the nozzle but right now it doesn't wait for cooldown. The setting for that is already there it is just not working as expected.

If the solution [here](https://github.com/supermerill/SuperSlicer/pull/2612) isn't satisfactory, point me to what would be better and I will update the PR

PierreMasselot1, 2022-04-01:
It's the exact same extruder (think about something like the MMU2),  and it doesn't wait for cooldown before filament extraction. That wait is critical to not have stringing in the single bowden tube that is shared by 5 filaments
[image]

supermerill, 2022-04-01:
@domesticatedviking Your though?

what i have with your 4 settings:
```
;WIDTH:1
G1 E-15 F6000
G1 E-55.3 F5400
G1 E-15.8 F2700
G1 E-7.9 F1620
M104 S100 ; FAST MODE <<<<<< SET THE TEMP
G1  Y139.900 
G1  X238.500  E5 F845
G1  X206.489  E-5 F911
G1  X238.500  E5 F977
G1  X206.489  E-5 F1043
G1  X238.500  E5 F1109
G1  X206.489  E-5 F1174
G1  X238.500  E5 F1240
G1  X206.489  E-5 F1306
M106 S255 ;Part fan on to cool hotend
M109 S100 ; FAST MODE <<<<<< WAIT FOR TEMP TO BE REACHED
M104 S240; FAST MODE
G1 E2 F2000
G1  Y140.000  F2400
```
the first line set the temp and the last one wait for it to be reach. It's in line with the "fast mode" you selected.

If I unselect the fast mode, I have 
```
;WIDTH:1
M106 S255 ;Part fan on to cool hotend
M109 S100 ; FAST MODE <<<<<< SET & WAIT THE TEMP
G1 E-15 F6000
G1 E-55.3 F5400
G1 E-15.8 F2700
G1 E-7.9 F1620
M104 S240; FAST MODE
G1  Y139.900 
G1  X238.500  E5 F845
G1  X206.489  E-5 F911
G1  X238.500  E5 F977
G1  X206.489  E-5 F1043
G1  X238.500  E5 F1109
G1  X206.489  E-5 F1174
G1  X238.500  E5 F1240
G1  X206.489  E-5 F1306
G1 E2 F2000
G1  Y140.000  F2400
```

I'm not convinced that the `M104 S240;` in the not-fast mode shouldn't be after the `G1  X206.489  E-5 F1306` like in the fast mode...

PierreMasselot1, 2022-04-01:
Agreed, the logic is not great there... This is stuff that I haven't  touched yet (the cooldown fix being the thing that has the most impact for people building the upgrade that I am working on). I think that fixing this could be in another issue and PR (and I will probably be working on it this weekend). The main thing that I am looking for now is getting that M109 S100 replaced by a M109 R100 for Marlin so that it actually waits (and so that I can point ppl to your release for the fix). If you want everything related to that feature fixed in the PR for this issue I can do it but would rather make incremental changes as I work on this feature.

supermerill, 2022-04-01:
> The main thing that I am looking for now is getting that M109 S100 replaced by a M109 R100 for Marlin so that it actually waits

It will be in the next nightly

schnello, 2022-04-05:
@supermerill 

A gen question to the temp change:
As i can see the sequence is:
Print Object
move to wipe tower
start Ramming
change Temp (M109 ...)

Makes this sense? After the Ramming der is a pressure in the nozzle and during waiting all filament in the nozzle is released to the wipe tower.

From my personal view it should me make more sense:
Print Object
move to wipe tower
change Temp (M109 ...)
start Ramming

or?

domesticatedviking, 2022-04-11:
My understanding of the goals of fast mode are pretty limited -- I coded it to Dave Shealey's spec because a) Skinnydip was originally his idea and b) he said in his hand-coded tests he was getting good results with it.  

My only real goal with toolchange temperatures was to ensure that I could have the hotend at a lower temperature by the time the filament was yanked out of the PTFE and up to the MMU2.     My reasoning was that at lower temperatures filament threads would stretch less and be more likely to snap off and remain in the melt zone.     I think that skinnydip moves (if enabled) should happen AFTER reaching the toolchange temperature -- the lower temperature will reduce stringing, and the skinnydip would melt off any string that remained.

A person could do that with a blocking temperature change (M109) immediately before the extraction (+ skinnydip move), but the cost of doing it that way would be time.    I found there were minimal issues with adding a nonblocking temperature change (M104) to begin dropping to the toolchange temperature while building the wipe tower, and then using the blocking M109 to ensure the target temperature had been reached just before extraction.

I've largely ignored Prusa's tip formation routines in my testing of the effects of toolchange temperatures -- I was never able to discern what changing those settings actually accomplished, so I don't really have an opinion about where they belong in the sequence.