supermerill/SuperSlicer #2616: New Release 2.4 Wall Thickness Increasing by itself

kpdcompany, 2022-04-01:
### Version
2.4

### Operating system type + version
Win10

### Behavior
See Picture, not sure how to describe it and not sure its influence, but wall thickness is extremely wrong.

Change Flow/extrusion width settings and check back to see if it changed.

#### Project File (.3MF) where problem occurs
_*Upload a Project File* (.3MF) `Plater -> Save Project as...`_
_You have to zip it (or rename it to .zip) to be able to drag& drop it here
[image]
[Wall Width Wrong.zip](https://github.com/supermerill/SuperSlicer/files/8400044/Wall.Width.Wrong.zip)
_

supermerill, 2022-04-01:
It's a shortcut, it has no effect by itself.

kpdcompany, 2022-04-01:
I'm sorry but I don't understand what it could possibly be making it such a high number though, and whats the point if it doesnt do anything?  Sorry if I dont understand.  Those large numbers will scare a lot of people on a setting that says wall thickness hahah

supermerill, 2022-04-01:
It's a new kind of setting, like the seam_position "corners", or the three settings in platter view: 
 * when set, they set other settings. 
 * What's displayed is the current view of the other settings.

That allows to present simplier settings to easy view.
For exemple, the xy compensation was split between inner and outer, and a shortcut setting can be created to be able to set both at the same time.

kpdcompany, 2022-04-01:
Ok so you would just set it for the total wall thickness you want? I do notice it changes a setting to "wall_thickness_lines = 2" notes, which seems its just setting back and forth the same settings, I'm just trying to understand the point, doesn't it just take it and adjust to the nearest walls needed? What variables does that Note variable set?

supermerill, 2022-04-01:
it set the perimeter widths.

the script code is in `yourconfdir/ui_layout/yourselectedlayout/print.as`
the definition is in `yourconfdir/ui_layout/yourselectedlayout/print.ui`
you can get `yourconfdir` from help->show conf dir
you select your `yourselectedlayout` in preferences->GUI->'settings layout & colors'

kpdcompany, 2022-04-01:
> it set the perimeter widths.
>
> the script code is in `yourconfdir/ui_layout/yourselectedlayout/print.as` the definition is in `yourconfdir/ui_layout/yourselectedlayout/print.ui` you can get [...]

Thank you, your the man!

pa0los, 2022-04-02:
FYI: I have also seen the same issue multiple times.
I'm using 0.6 nozzle - maybe it's related?

Tuxomx, 2022-04-13:
Hi. Just update to 2.4.58.2 and found this issue too. If this number reflects other parameters as stated before, then why it is changing? In the following screenshots never changed a setting, just took the image after starting the program. Even after saving that parameter to 0 (as was the setting with previous versions) the vehavior continues. Now, if this refect other settings, does those settings are changing too? Does shell thickness change too? This is important to me because currently I am doing lots of calibration cubes for statistical evaluation and teach my students how to prove some things with ANOVA, we are measuring shell thickness changing some parameters. 

SALUDOS
ALVARO

[image]
[image]
[image]
[image]

supermerill, 2022-04-13:
@Tuxomx
can you test a nightly?

Tuxomx, 2022-04-13:
> @Tuxomx can you test a nightly?

Sure. Just now downloaded one from tree days ago, 2.4.58.3, this one does not have that option, here is how it looks in my computer.

[image]

Tuxomx, 2022-04-14:
> it's only in easy mode.

Oh, Well, I did some changes, but found this. I am using fedora 35 with mate desktop (GTK3).

[image]

I think should be radio buttons or no buttons at all since is a menu entry. After selecting only the easy mode it gets like this:
[image]

If no radio buttons I would place a small label indicating the mode.

supermerill, 2022-04-14:
The simple/advanced/expert aren't mode anymore. They're now tags.
If you just want to select one, just click on the colored ones in the main GUI. The menu entry now lets you choose the combination you want to use.

> So I have no idea why wall thickness is .76 with three perimeters when nozzle (in advanced configuration) es .4. Regarding the menu, I need to open it twice to change correctly to the desired configuration mode.

it's just a number. For the number of lines a wall uses, it get the number from the custom variable wall_thickness_lines. If it's not present, it assumes 2 lines/1 loop (so the number you get from the little text just below).

I have too many reports about this quick-setting thing. I have to change the label & the tooltip, and maybe always show 0?

What about `Recommended min wall thickness` as the new label?
and change the tooltip from 
`Change the perimeter extrusion width to ensure that there is an exact number of perimeters for this wall value. It won't put the width below the nozzle diameter, and up to double the size of the nozzle.` 
to
`This quick setting is a tool to set the extrusion width easily to have walls with a perfect number of perimeters. It will show you the last value set or the min wall thickness otherwise. When set, it will choose a perimeter width so there can be an exact number of perimeter lines to fill this thickness. It won't put the width below the nozzle diameter, and up to double the size of the nozzle. The number of lines used is stored in the print custom variables (visible is advanced and expert mode).`

Tuxomx, 2022-04-15:
> The simple/advanced/expert aren't mode anymore. They're now tags. If you just want to select one, just click on the colored ones in the main GUI. The menu entry now lets you choose the combination you want to use.
Yes, understood, I am able to look at the buttons, althoug menu entries seem a bit confusing.

> it's just a number. For the number of lines a wall uses, it get the number from the custom variable wall_thickness_lines. If it's not present, it assumes 2 lines/1 loop (so the number you get from [...]
I see, I understood that, although math didn't worked for me, let me check, In any casi, looks like I am getting good prints, still hadn't tested printing with nightly.

> 
> I have too many reports about this quick-setting thing. I have to change the label & the tooltip, and maybe always show 0?
Yep, I think showing 0 by default like before. Maybe reports are because we are used to leave it alone and with zero value. Also, I think it is not bad idea if it shows the actual value. For me, what make me come here and talk was the fact that number was changing every time I started the program.
> What about `Recommended min wall thickness` as the new label? and change the tooltip from `Change the perimeter extrusion width to ensure that there is an exact number of perimeters for this wall [...]

I like your tooltip very much, it states very clearly what is happening.

 If you feel like I can help you testing some settings by  printing something don't hesitate to ask me and as soon as possible I can test nightly and print something not being too large.

closed 2024-02-11 by supermerill