supermerill/SuperSlicer #2619: External Seam? Question

jjj03, 2022-04-02:
I'm making cookie cutters and need the inside perimeter to be seamless. Is there a way to set an aligned external perimeter instead?

pa0los, 2022-04-02:
By _inside perimeter_ you probably mean **external perimeter** but from the inside of the shape.
There will always be a seam on the external perimeter somewhere. You can only hide it in a corner or tune your profile/printer settings so that it's less visible, even on round shapes.

jjj03, 2022-04-02:
I figured it out. Cura has the hide/expose seam which I always picked expose so it was on the outside of the design. But in super slicer, I just put a much larger value into seam gap, which produced a gap in the slicer view, but it's perfect once printed.