supermerill/SuperSlicer #2620: 2.4.58.1/2.4.58.2 Crash on startup

lordIcocain, 2022-04-02:
### What happened?

Got crash on startup in last two version of 2.4
2.4.58.0 working
2.4.58.1 can start only when AppData/SuperSlicer empty 
2.4.58.2 crash in any case (still creating files in AppData/SuperSlicer)
[image]

### Project file & How to reproduce

[image]

### Version

2.4.58.1, 2.4.58.2

### Operating system

win10 

### Printer model

Any

lordIcocain, 2022-04-02:
Hello, just tested night build, no problem with startup. Thank you.
Still get some errors, but not saw any problems in work.
[image]

supermerill, 2022-04-03:
can you help me to debug these errors?
As long as they're here, you won't have quick settings and some settings will be missing.

can you try to add a "configuration" directory next to your superslicer.exe file, and report if it still fail?

lordIcocain, 2022-04-03:
[image]

lordIcocain, 2022-04-03:
Last night build with folder "configuration"
[image]
and without
[image]

lordIcocain, 2022-04-05:
[image]

lordIcocain, 2022-04-06:
Seems like it trying load ui_script files what actually does not exist.
filament.as
sla_material.as
printer.as
These files were not required in previous versions, they were not in the settings directory and there were no errors that these files were not found.
The program began to request these files in a nightly update that fixed bugs with Cyrillic(UTF) path.

supermerill, 2022-04-07:
> Seems like it trying load ui_script files what actually does not exist.

I know, and it's indenteded. There is currently no scripted widget on these tab, and so i didn't created the scripted file,n as they aren't needed. You can see that now that I added logs, but it's completly harmless.

do you still don't have the quick setting on the platter tab?

lordIcocain, 2022-04-08:
I'm confused. I dont get any problems in program working. The quick settings window opens and works as it should.

supermerill, 2022-04-08:
> The quick settings window opens

the "quick settings" are the three widgets here
[image]
and if the script fail, you can only have the "infill" one (as it's the only one that isn't scripted)

lordIcocain, 2022-04-08:
[image]
it ok too

supermerill, 2022-04-08:
great!

closed 2022-04-17 by lordIcocain