supermerill/SuperSlicer #2621: Multi-Material Painting?

techni-tech, 2022-04-02:
With the 2.4 merge I expected that Superslicer would also feature multi-material painting, even more since it is currently the best slicer for people with a toolchanger and this feature is HUGE for us.

But why is this feature (still?) missing?

Apologies if this question came up before but I searched the patch notes and could not find anything. In fact, the notes of the "Beta 2.4: acceleration edition" say that now you have "ALL the new stuff from" PrusaSlicer 2.4.1 but this obviously is not the case.

Is this feature still too buggy to enable it? Is it not open-source? Or what is the reason?

Pavulon87, 2022-04-02:
The painting works from the first beta versions
[image]

supermerill, 2022-04-02:
You have to setup a printer with multiple extruders for it to appear, maybe that's what confuse you?

techni-tech, 2022-04-03:
Ah I see what the issue is (not the setup with multiple extruders). 

So you recently released a new beta for 2.4 and just before that there was another 2.3.57.12 version (Gravel solid edition), which does not include the painting feature but was the one I tested and mistook for a 2.4 release. My bad...

Just tested the painting feature in SuperSlicer and seems to work fine!

closed 2022-04-03 by techni-tech