supermerill/SuperSlicer #2622: Top layer quality

aceycv8, 2022-04-02:
### Version
2.3.57 
### Operating system type + version
Windows 10

### Behavior
i'm getting uneven top layer quality on my prints i'm not sure whats causing this, i'm using 4 top layers. my flow rate is set to 98% for the top layer in hopes it would help with smoothness but as you can see it's not uniform at all. any suggestions would be appreciated - using a .4 nozzle 

thank you

[image]

supermerill, 2022-04-03:
are you using the monotonic pattern for th top?

aceycv8, 2022-04-03:
ahh no i had rectilinear selected - i will try monotonic

thank you supermerill