supermerill/SuperSlicer #2623: Multimaterial print temp problems

hagan66, 2022-04-02:
### What happened?

I set up a very basic multi material print and noticed that temps drop to zero after material change. 

Also when using Fast mode and Use part cooling fan to cool hotend at the filament change the filament fan speeds were ignored.

### Project file & How to reproduce

[Fishy edition Calibration cube.3mf.zip](https://github.com/supermerill/SuperSlicer/files/8402312/Fishy.edition.Calibration.cube.3mf.zip)

### Version

2.4.58.2

### Operating system

Macos 11.6.5

### Printer model

Prus MK2.5S MMU2S

supermerill, 2022-04-02:
Looking at the gcode, it seems good. maybe a gui-preview bug or gcode parser problem.

hagan66, 2022-04-02:
Ok, what made me look was that I started a print and I noticed the fan was off, so that was real.

supermerill, 2022-04-02:
understood. It track the temp per extruder, and the next extruder doesn't receive a temp command after  the 'T1'.
But it's a single extruder mmu, so only one temp.
Will fix the gcode reader.

supermerill, 2022-04-02:
I also fixed the fan speed bug (which is real, in the gcode)

hagan66, 2022-04-02:
Great, thanks! 👍🏻

Skickat från min iPhone