supermerill/SuperSlicer #2624: Can't open recent project

Fuechsl, 2022-04-02:
### Version
2.3.57.12

### Operating system type + version
MacOS 12.3.1

### Behavior
* When opening a recent project (saved last week) I get the error "Failed loading file xxx due to an invalid configuration. Invalid value provided for parameter fill_ pattern:"
[image]

#### Project File (.3MF) where problem occurs
[Griff erweitert2 ASA 90 grad.3mf.zip](https://github.com/supermerill/SuperSlicer/files/8402388/Griff.erweitert2.ASA.90.grad.3mf.zip)
(I had to zip the 3mf file since github doesn't support the file type)

supermerill, 2022-04-02:
duplicate of #1983 #997
the 2.4 version can "repair" these kind of issue.
The problem lie when it was saved.
I'm trying to understand why it fail to save the value, for now i have no clue.

supermerill, 2022-04-02:
[2624_Griff erweitert2 ASA 90 grad.3mf.zip](https://github.com/supermerill/SuperSlicer/files/8402501/2624_Griff.erweitert2.ASA.90.grad.3mf.zip)

Fuechsl, 2022-04-02:
Thanks. But 2.4 is not yet available?

----------------------------------
Martin Lacher
Pilatusstrasse 32
CH-6003 Luzern
+41 (0)79 778 29 59

supermerill, 2022-04-02:
it's still in beta. 
you can get it in the release page.