supermerill/SuperSlicer #2625: Choco package not updated to 2.3.57.12

reapola, 2022-04-02:
### What happened?

Superslicer via Chocolatey is still on 2.3.57.11 - can this be updated please?

### Project file & How to reproduce

`C:\Windows\system32>choco upgrade superslicer
Chocolatey v0.12.1
Upgrading the following packages:
superslicer
By upgrading, you accept licenses for the packages.
superslicer v2.3.57.11 is the latest version available based on your source(s).

Chocolatey upgraded 0/1 packages.
 See the log for details (C:\ProgramData\chocolatey\logs\chocolatey.log).

Enjoy using Chocolatey? Explore more amazing features to take your
experience to the next level at
 https://chocolatey.org/compare`

### Version

2.3.57.11

### Operating system

Windows 10

### Printer model

N/

supermerill, 2022-04-03:
the validation is pending.

mmirate, 2022-04-03:
Meantime, if you are impatient and trustful, you can specify the version manually `choco upgrade superslicer --version 2.3.57.12` to install a chocolatey package version that has not yet passed validation.