supermerill/SuperSlicer #2626: Coasting or an Alternative pls

inspectionsbybob, 2022-04-02:
Coming from Cura, I am used to being able to tune then"coasting" to get rid of the blob at layer changes.  I have tweaked my retraction so is much (much) better, but I still have a small artifact on layer changes.

I am not sure if it is the end one layer, or the beginning of the next, I may play with a negative de-retract to see if it the beginning of the next line... but coasting is still a requested feature to eliminate any remnant of a blob at the end of a layer...

I looked at the resolution as a possibility, but I am only seeing ONE blob per layer, and even if I user ArcWelder, which (greatly) simplifies curved layers, I am still seeing blobs... anyway my controller is a new one so it should have no problem with keeping up.  Only time I have see it have issues was with a very dense Gyroid infill, the front panel got unresponsive... not seeing that with blobs...

Anyway, looking for a way to get rid of the last remnants of Blogs/Zits on my prints.

NOTE: Any fix/feature should be in the FILAMENTs section as it is VERY filament dependent.  Some filaments blog regardless of temperature/retraction, others come out beautiful regardless...(carbon fiber). Silky filaments seem the worst, or it just more noticeable...

supermerill, 2022-04-02:
Filament->overrides->seam gap

strayr, 2022-04-15:
set up linear advance/pressure advance on your printer, most firmware/hardware combinations are capable of this and coasting is an attempt to solve the deceleration bulge that pressure advance solves along with acceleration underextrusion.

inspectionsbybob, 2022-04-26:
Ok, I have been trying to get rid of the Seam Gap or Z-line using everything that is even close to that section.

No Joy...

As you can see on my test cylinder, I have TWO bumps that make up the Z-seam. I assume a start and a stop.  No matter what I try, they stay. Either I have something turned on that is negating my work, or this is as good as it gets. All of my work and all I have managed to do is spread the bumps apartq a bit.

[image]

I am trying to print a "Print in Place" globe with some sliding parts.  That little bump was enough to fuse a sliding part to the wall...

I looked into "Linear Advance" and unfortunately my printer uses 2208s, so I can't use LA without recompiling which is W a y  past  what I can do...

and here is the actual print that failed...  I can free up everything but the z-line.
 
[image]

Any help you can give would be appreciated.

inspectionsbybob, 2022-04-26:
Here is my Project file...
ALSO, in looking through the open issues, there is one about seams on Cylinders. Is this related?

[Test Cylinder.zip](https://github.com/supermerill/SuperSlicer/files/8567445/Test.Cylinder.zip)
?

inspectionsbybob, 2022-10-11:
On reading the help for that,  it appears that is for FILLING gaps at the
z-seam.  I am having the OPPOSITE  problem in that I have a BLOB not a gap.

-- 
-----------------------------------------------------------------------------------
Need a quick answer? Call our booking line at (301) 208-8289, or book your
inspection on our website 24/7.
Don't wait for a response as someone else may get the spot you want... call
Today!
-----------------------------------------------------------------------------------
Bob Sisson, ACI, BVI
Inspections by Bob, LLC
www.inspectionsbybob.com
***@***.***
MD Lic#29666; ASHI Member #212016

closed 2023-04-24 by inspectionsbybob