supermerill/SuperSlicer #2627: Per-object settings are removed when canceling new setting dialog

pa0los, 2022-04-02:
### What happened?

It took me a while to discover why/when my per-object settings are getting removed.
On the example of custom "Width Flow" settings assigned to an object:
[image]
If I now open the dialog to add another Width Flow setting, I can see these 2 options checked:
[image]
But when I hit Cancel in this dialog box, the 2 settings that I already had are getting removed:
[image]

### Project file & How to reproduce

Not related to a particular project.
Just add a per-object setting, open the "Add settings" dialog box again and hit cancel.

### Version

2.4.58.2

### Operating system

Windows 10

supermerill, 2022-04-02:
If I change the button label to "Apply" and "Remove All", is it okay?
I'll see if I can add a third.

pa0los, 2022-04-02:
Oh I see - this is how it's supposed to work...
Renaming the buttons like you proposed would make it clear I think.

supermerill, 2022-04-03:
It took me some hours, but I have fixed it.
* Apply apply the selection, like the ok
*  Remove all (new) now erase the selection and apply.
* Cancel now don't touch the items

In the codebase, it was a bit strange. some code say "cancel don't erase" and then the caller erase everything...

supermerill, 2022-04-03:
"Remove All" or "Clear" ?

pa0los, 2022-04-03:
If you ask me, I would only leave "apply" and "cancel" buttons.
Removing is easy enough and I can't imagine having this many settings to justify the "remove all / clear" button.
But "clear" is also ok.

closed 2024-02-11 by supermerill