supermerill/SuperSlicer #2632: [BR] XY Compensation make steps in the sliced model

ldiegos, 2022-04-03:
### What happened?

When XYCompensation has different value from 0 in either Outer or Inner make steps in the sliced model.
Following the explanation on the popup message do not help or correct the step in the design, because i have some hollowed letters in the model in Vertical plane:

I think that susl have issues to identify which type of perimeter is when goes to the inner perimeters after turn inside the letter. 

The printer with susl gave me some differences when printing in the final result and i need to print the same model in several printers, so i try to adapt the XYcompensation to have the same result in this one, like the others, without modeling again with a new tolerance only for this one.

Sliced model with letters:
[image]

Different amounts:
[image]
Result:
[image]

Same amounts:
[image]
Result:
[image]

Only outer:
[image]
Result:
[image]

Only inner:
[image]
Result:
[image]

No XYCompensation:
[image]

Thank you so much in advance SuperMerill
Luis.

### Project file & How to reproduce

[XCortechXT301-XCortech-XT301-MK2-Base-WithLetters-Rabbit-discodecki-Part012.zip](https://github.com/supermerill/SuperSlicer/files/8404229/XCortechXT301-XCortech-XT301-MK2-Base-WithLetters-Rabbit-discodecki-Part012.zip)

### Version

2.3.57.7; 2.3.57.11

### Operating system

Windows 10

### Printer model

Geetech A10 with a ED3V6

supermerill, 2022-04-03:
duplicate of #2193
also #612

ldiegos, 2022-04-03:
Thank you SuperMerrill i didn't find it /search in deep, sorry.
So i'm screw... because there is not a solution... issue 612 and 2193 seems to be so old without improvement

ldiegos, 2022-04-04:
Good morning again... just understand whats the meaning of "same value":
[image]

If i put the same value, the XYcompensation works equal to PrusaSlicer2.4 and there is no steps in the sliced model. I can work with it, my problem was the measure of the internal circle, but the external circle to be a little bit more thin, is not a huge issue, at least for this print.

[image]

I will save the preset just for making tubes/covers.

Thanks a lot and waiting to have a improvement about XYCompesantion
Luis.