supermerill/SuperSlicer #2633: The new version sucks!

AntoszHUN, 2022-04-03:
I made the Hungarian language for this cretan program. You have released a new version and the Hungarian language is not included! I spent several days on it unnecessarily. Now I can start all over again. This is not normal. Of course, there are other unnecessary languages in it. I WANT TO HUNGARIAN!

I got angry!

closed 2022-04-03 by AntoszHUN

supermerill, 2022-04-03:
I'm sorry but i'm a bit out of the loop.
There is hungarian in the latest version.
Maybe I made a mistake?
Instead of ranting, please tell me where are my mistake so I can fix them.

AntoszHUN, 2022-04-03:
> 

> I'm sorry but i'm a bit out of the loop. There is hungarian in the latest version. Maybe I made a mistake? Instead of ranting, please tell me where are my mistake so I can fix them.

I do not care anymore. I'll do a meaningful translation then it will be in it. But the latest version is not sure.