supermerill/SuperSlicer #2634: Failed to load image from file (SuperSlicer icon) upon startup

bradvaughan, 2022-04-03:
### What happened?

Upon startup, the following error is displayed.  Clicking "OK" does allow the app to continue starting and operation is as expected thereafter.  I verified there is no such file in resources\splashscreen, but there wasn't in 2.3.57.11 either.

[image]

### Project file & How to reproduce

Open SuperSlicer, wait for the splash screen to disappear.  This error message appears immediately afterwards.

### Version

2.4.58.2

### Operating system

Windows 10

### Printer model

N/A

bradvaughan, 2022-04-03:
This error can be resolved by changing the splash screen image from "Default" to "Icon":

Configuration -> Preferences -> General -> Splash Screen -> Splash screen image:  Icon

supermerill, 2022-04-03:
I guess it was a bad entry in the color.ini that was in your conf dir.

bradvaughan, 2022-04-03:
Non repro

closed 2022-04-03 by bradvaughan

jangrewe, 2022-05-16:
Can confirm both the issue and the fix. Happened when updating stable to latest pre-release.