supermerill/SuperSlicer #2635: Autospeed produces pathologically slow speeds for some features when layer time is low

matti125, 2022-04-03:
### What happened?

Using autospeed sometimes produces pathologically slow speeds for internal perimeters. It seems to occur when the layer time is rather low. Speeds drop down to values like 1mm/s, while perimeter speed is >100mm/s. The resulting near-stall produces poor surface quality and wastes time, layer time can be several tens of minutes. In the worst case, the print gets ruined, as at least with my lgx lite/rapido combination the filament stopped feeding after such an event. (I am not sure why that happened, did not debug, but suspect a clog or a filament snap-off)

### Project file & How to reproduce

[slow.3mf.zip](https://github.com/supermerill/SuperSlicer/files/8404572/slow.3mf.zip)
[slow_layer.3mf.zip](https://github.com/supermerill/SuperSlicer/files/8404600/slow_layer.3mf.zip)

### Version

2.3.57

### Operating system

macos 12.2.1

### Printer model

Ratrig vcore3, LGX lite, Rapido hotend

supermerill, 2022-04-03:
It doesn't lower than the min speed set in the filament cooling.
I don't know what's wrong with them.

matti125, 2022-04-03:
Ok, then something differs between your setup and mine. Seems your top speeds are lower than mine (44mm/s vs 140mm/s)) How can that be debugged? Using the same slow_layer.3mf gets me this:
[image]

supermerill, 2022-04-03:
You are using the 2.3.57.12, right?

I don't know how you can go that fast, as the project has a 10sec min layer time:
[image]

matti125, 2022-04-03:
Hmm, no, this is still 2.3.57.4.  But I cannot understand the sub 1s times either. The min speed is 15mm/s and the object is abt 14mm wide, so with 2 perimeters I believe we should be at around 4 secs as the worst case. But let me try with a newer version.

supermerill, 2022-04-03:
> this is still 2.3.57.4

so problably a bug that has been fixed since then.

matti125, 2022-04-03:
Issue not visible in 2.3.57.12. So yes, probably a bug that got fixed between .4 and .12. Thanks!